DoosanRobotics/doosan-robot2 #2: [ERROR] [dsr_control_node2-4]: process has died [pid 4885, exit code -11, cmd ...]

MartinCicmanec, 2021-01-14:
Hello, I have problems running both the virtual mode robot and the real robot. I installed ROS2 using the official guide and then followed the install guide exactly as it is in the README.md in this repository. Colcon built all packages successfully, but I got some warnings during the compilation of dsr_control2 :

`[Processing: dsr_control2]                                         
--- stderr: dsr_control2                                            
In file included from /home/robot/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:215,
                 from /home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/../../../common2/include/DRFL.h:360:6: warning: extra ‘;’ [-Wpedantic]
  360 |     };
      |      ^
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In member function ‘int dsr_control2::DRHWInterface::MsgPublisher_RobotState()’:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:474:60: warning: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  474 |         memcpy(&m_stDrState, &g_stDrState, sizeof(DR_STATE));
      |                                                            ^
In file included from /home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:399:16: note: ‘struct DR_STATE’ declared here
  399 | typedef struct {
      |                ^
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In member function ‘int dsr_control2::DRHWInterface::MsgPublisher_JointState()’:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:617:60: warning: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  617 |         memcpy(&m_stDrState, &g_stDrState, sizeof(DR_STATE));
      |                                                            ^
In file included from /home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:399:16: note: ‘struct DR_STATE’ declared here
  399 | typedef struct {
      |                ^
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:618:26: warning: statement has no effect [-Wunused-value]
  618 |         msg.header.stamp.sec;
      |         ~~~~~~~~~~~~~~~~~^~~
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static void dsr_control2::DRHWInterface::thread_subscribe(rclcpp::Node::SharedPtr)’:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:650:66: warning: unused parameter ‘nh’ [-Wunused-parameter]
  650 |     void DRHWInterface::thread_subscribe(rclcpp::Node::SharedPtr nh)
      |                                          ~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In constructor ‘dsr_control2::DRHWInterface::DRHWInterface(rclcpp::Node::SharedPtr&)’:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:923:52: warning: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
  923 |         memset(&g_stDrState, 0x00, sizeof(DR_STATE));
      |                                                    ^
In file included from /home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:399:16: note: ‘struct DR_STATE’ declared here
  399 | typedef struct {
      |                ^
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:925:52: warning: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
  925 |         memset(&m_stDrState, 0x00, sizeof(DR_STATE));
      |                                                    ^
In file included from /home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:399:16: note: ‘struct DR_STATE’ declared here
  399 | typedef struct {
      |                ^
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In member function ‘virtual hardware_interface::return_type dsr_control2::DRHWInterface::init()’:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szController’ [-Wmissing-field-initializers]
 1042 |             SYSTEM_VERSION tSysVerion = {'\0', };
      |                                                ^
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szInterpreter’ [-Wmissing-field-initializers]
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szInverter’ [-Wmissing-field-initializers]
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szSafetyBoard’ [-Wmissing-field-initializers]
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szRobotSerial’ [-Wmissing-field-initializers]
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szRobotModel’ [-Wmissing-field-initializers]
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szJTSBoard’ [-Wmissing-field-initializers]
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1042:48: warning: missing initializer for member ‘_SYSTEM_VERSION::_szFlangeBoard’ [-Wmissing-field-initializers]
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1081:26: warning: variable ‘eTargetSystem’ set but not used [-Wunused-but-set-variable]
 1081 |             ROBOT_SYSTEM eTargetSystem = ROBOT_SYSTEM_VIRTUAL;
      |                          ^~~~~~~~~~~~~
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In member function ‘void dsr_control2::DRHWInterface::trajectoryCallback(moveit_msgs::msg::DisplayTrajectory_<std::allocator<void> >::SharedPtr) const’:
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1204:20: warning: unused variable ‘sn_cnt’ [-Wunused-variable]
 1204 |         static int sn_cnt = 0;
      |                    ^~~~~~
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1241:15: warning: unused variable ‘preTargetTime’ [-Wunused-variable]
 1241 |         float preTargetTime = 0.0;
      |               ^~~~~~~~~~~~~
/home/robot/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1243:15: warning: unused variable ‘targetTime_sec’ [-Wunused-variable]
 1243 |         float targetTime_sec = 0.0;
[... 54 lines cut ...]
[INFO] [1610620113.208160892] [dsr_hw_interface2]: model = m1013
[INFO] [1610620113.208165495] [dsr_hw_interface2]: gripper = none
[INFO] [1610620113.208169244] [dsr_hw_interface2]: name_space is dsr01, m1013
[INFO] [1610620113.208173347] [dsr_hw_interface2]: constructed
[INFO] [1610620113.375226917] [dsr_hw_interface2]: [dsr_hw_interface2] init() ==> setup callback fucntion
[INFO] [1610620113.375261640] [dsr_hw_interface2]: joint_name = joint1
[INFO] [1610620113.375283442] [dsr_hw_interface2]: joint_name = joint2
[INFO] [1610620113.375290947] [dsr_hw_interface2]: joint_name = joint3
[INFO] [1610620113.375295301] [dsr_hw_interface2]: joint_name = joint4
[... cut ...]
Also, when I run `rosdep install --from-paths src --ignore-src --rosdistro foxy -r -y` I get an error saying `dsr_gazebo2: Cannot locate rosdep definition for [sdf]`. I tried googling it but could not find how to fix this dependency. However, I think this error is unrelated to the dsr_control2 issue.

I am not very skilled in C++, but I had an application running in ROS1 and I would like to transfer it to ROS2. 
It should not be caused by faulty config, because I am using all the example. Any help would be greatly appreciated.

doosan-robotics, 2021-04-13:
The doosan robot2 ros package has been updated.
After updating the package, please follow the process in Readme.md and try again.

MartinCicmanec, 2021-04-13:
Thank you. I will test this ass soon as I can and report the result here.

Serafadam, 2021-04-14:
Hi, I've just cloned this repository for some tests, and I seem to be having the same problem when running the examples, I also tried to run DRCF manually.

doosan-robotics, 2021-04-15:
I have inserted the DRCF auto-run script in each launch.py ​​file that currently exists in the dsr launcher2 package.
If it does not run normally, run the DRCF64 binary file in common2/bin/DRCF with sudo privileges.

Serafadam, 2021-04-15:
This is output after running `ros2 launch dsr_launcher2 single_robot_rviz.launch.py`

``` bash
ros2 launch dsr_launcher2 single_robot_rviz.launch.py                                                                                        
[INFO] [launch]: All log files can be found below /home/username/.ros/log/2021-04-15-09-06-28-466525-FTKRL5771-4892
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sh-1]: process started with pid [4895]
[INFO] [static_transform_publisher-2]: process started with pid [4898]
[INFO] [robot_state_publisher-3]: process started with pid [4901]
[INFO] [rviz2-4]: process started with pid [4903]
[INFO] [dsr_control_node2-5]: process started with pid [4906]
[sh-1] Run Emulator of the Doosan Robot Controller
[sh-1] dirname: /home/username/coding/doosan_ws/install/common2/share/common2/bin/DRCF/run_drcf.sh
[sh-1] server_port: 
[sh-1] robot model: 
[sh-1] ARCH: 64-bit
[static_transform_publisher-2] [INFO] [1618470388.657164554] [static_transform_publisher]: Spinning until killed publishing transform from 'base' to 'base_0'
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_0 had 1 children
[robot_state_publisher-3] Link link1 had 1 children
[robot_state_publisher-3] Link link2 had 1 children
[robot_state_publisher-3] Link link3 had 1 children
[robot_state_publisher-3] Link link4 had 1 children
[robot_state_publisher-3] Link link5 had 1 children
[robot_state_publisher-3] Link link6 had 0 children
[robot_state_publisher-3] [INFO] [1618470388.660773502] [robot_state_publisher]: got segment base_0
[robot_state_publisher-3] [INFO] [1618470388.660880373] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1618470388.660896999] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1618470388.660905650] [robot_state_publisher]: got segment link3
[robot_state_publisher-3] [INFO] [1618470388.660912574] [robot_state_publisher]: got segment link4
[robot_state_publisher-3] [INFO] [1618470388.660919891] [robot_state_publisher]: got segment link5
[robot_state_publisher-3] [INFO] [1618470388.660927077] [robot_state_publisher]: got segment link6
[robot_state_publisher-3] [INFO] [1618470388.660934582] [robot_state_publisher]: got segment world
[dsr_control_node2-5] [INFO] [1618470388.668333796] [dsr_control_node2]: g_node = 0x0x7ffc76a32db0
[dsr_control_node2-5] [INFO] [1618470388.668443428] [dsr_control_node2]: rate is 100
[dsr_control_node2-5] [INFO] [1618470388.668499940] [dsr_control_node2]: controller_manager is updating!
[dsr_control_node2-5] [INFO] [1618470388.668542965] [dsr_hw_interface2]: name = dsr01
[dsr_control_node2-5] [INFO] [1618470388.668557845] [dsr_hw_interface2]: model = m1013
[dsr_control_node2-5] [INFO] [1618470388.668561583] [dsr_hw_interface2]: gripper = none
[dsr_control_node2-5] [INFO] [1618470388.668564815] [dsr_hw_interface2]: name_space is dsr01, m1013
[dsr_control_node2-5] [INFO] [1618470388.668568631] [dsr_hw_interface2]: constructed
[dsr_control_node2-5] [INFO] [1618470388.707567665] [dsr_hw_interface2]: [dsr_hw_interface2] init() ==> setup callback fucntion
[dsr_control_node2-5] [WARN] [1618470388.707620189] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[dsr_control_node2-5] [INFO] [1618470388.707595587] [dsr_hw_interface2]: joint_name = joint1
[dsr_control_node2-5] [INFO] [1618470388.707787067] [dsr_hw_interface2]: joint_name = joint2
[dsr_control_node2-5] [INFO] [1618470388.707828264] [dsr_hw_interface2]: joint_name = joint3
[dsr_control_node2-5] [INFO] [1618470388.707843358] [dsr_hw_interface2]: joint_name = joint4
[dsr_control_node2-5] [INFO] [1618470388.707859511] [dsr_hw_interface2]: joint_name = joint5
[dsr_control_node2-5] [INFO] [1618470388.707871397] [dsr_hw_interface2]: joint_name = joint6
[dsr_control_node2-5] [INFO] [1618470388.707881425] [dsr_hw_interface2]: INIT@@@@@@@@@@@@@@@@@@@@@@@@@
[dsr_control_node2-5] [INFO] [1618470388.707888165] [dsr_hw_interface2]: init() ==> arm is standby
[dsr_control_node2-5] [INFO] [1618470388.707904313] [dsr_hw_interface2]: host = 127.0.0.1
[dsr_control_node2-5] [INFO] [1618470388.707910746] [dsr_hw_interface2]: port = 12345
[dsr_control_node2-5] [INFO] [1618470388.707916184] [dsr_hw_interface2]: command = 1
[dsr_control_node2-5] [INFO] [1618470388.707921506] [dsr_hw_interface2]: mode = virtual
[dsr_control_node2-5] [INFO] [1618470388.707926615] [dsr_hw_interface2]: host 127.0.0.1, port=12345 bCommand: 1, mode: virtual
[ERROR] [dsr_control_node2-5]: process has died [pid 4906, exit code -11, cmd '/home/username/coding/doosan_ws/install/dsr_control2/lib/dsr_control2/dsr_control_node2 --ros-args -r __node:=dsr_control_node2 --params-file /tmp/launch_params_0t7dsgfi --params-file /tmp/launch_params_1vfz9yke --params-file /tmp/launch_params_e7csxitm --params-file /tmp/launch_params_e4_nhmzy --params-file /tmp/launch_params_0vzcymko --params-file /tmp/launch_params_h2qtx0uo --params-file /tmp/launch_params_tgx2tbef --params-file /tmp/launch_params_37ajk6dn --params-file /tmp/launch_params_gxzx7vrl --params-file /tmp/launch_params_9mujqjxr'].
[... cut ...]
[sh-1] #SAFETY MODE: SAFETY_MODE_INITIALIZE -> SAFETY_MODE_MANUAL
[sh-1] EtherCAT Operation mode start
[sh-1] RUN CONTROL : 1
[sh-1] Key Command : 9
[sh-1] Key Command : 27
[sh-1] 
[sh-1] STOP ROS service...
[sh-1] The Emulator is finished!
[sh-1] Final OnTask Closed...
[ERROR] [sh-1]: process has died [pid 4895, exit code -2, cmd 'sh /home/username/coding/doosan_ws/install/common2/share/common2/bin/DRCF/run_drcf.sh'].
[INFO] [rviz2-4]: process has finished cleanly [pid 4903]
```
If I run DRCF manually, the problem persists

bleut, 2021-04-20:
I have same problem when I try to run <code> ros2 launch dsr_launcher2 single_robot_rviz.launch.py<\code> 

<code>[ERROR] [dsr_control_node2-5]: process has died [pid 57772, exit code -11, cmd '/home/hmdo/ros2_ws/install/dsr_control2/lib/dsr_control2/dsr_control_node2 --ros-args -r __node:=dsr_control_node2 --params-file /tmp/launch_params_whumkhe1 --params-file /tmp/launch_params_fhyzmk30 --params-file /tmp/launch_params_ow3urbmi --params-file /tmp/launch_params_k80q2li1 --params-file /tmp/launch_params_6ncfob1b --params-file /tmp/launch_params_yilajx76 --params-file /tmp/launch_params_qi_az6_v --params-file /tmp/launch_params_2r_i37fe --params-file /tmp/launch_params_v0b7v8j_ --params-file /tmp/launch_params_6r_xl35y'].<\code>

Any ideas to fix this problem?

doosan-robotics, 2021-04-26:
@Serafadam @bleut 
Manually run DRCF in one terminal, and run the launch file using the command ros2 launch dsr_launcher single_robot_rviz.launch in the other terminal. Do you have the same problem?
Or delete all files in the workspace's build and install directory and try to build again.

MartinCicmanec, 2021-04-26:
I have managed to successfully build the packages in docker and it works. I had to install "libpoco-dev" and I see it has been added to README.md today so I would consider this issue fixed. Thank you.

closed 2021-04-26 by MartinCicmanec