DoosanRobotics/doosan-robot2 #3: [ERROR]: the following packages/stacks could not have their rosdep keys resolved to system dependencies: dsr_gazebo2:

bleut, 2021-04-12:
Hello, I have problems in installing doosan-robot2 package.
I'm trying to install doosan-robot2 package for ros2 according to the build procedure described in github.
 
When I run <code>rosdep install --from-paths src --ignore-src --rosdistro foxy -r -y</code> 
I got an error saying <code>ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies: dsr_gazebo2: Cannot locate rosdep definition for [sdf]</code>

Pls help me to fix this problem.

doosan-robotics, 2021-04-13:
The wrong package was contained in package.xml. There is no problem in practical use.

Thank you.

bleut, 2021-04-27:
I think this issue was fixed. Thank you for your help.

closed 2021-04-27 by bleut