DoosanRobotics/doosan-robot2 #4: [ERROR] build problem dsr_gazebo2 and dsr_control2

whitedoll, 2021-04-12:
Hellow, I'm using Ubuntu 20.04 and I installed ros2 foxy via Debian package
I will inform you of an error that occurred when I installed the Doosan Robot package.
How can I install the package without errors?

1)
--- stderr: gazebo_ros2_control
fatal error: hardware_interface/base_interface.hpp: No such file or directory
   27 | #include "hardware_interface/base_interface.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/gazebo_hardware_plugins.dir/build.make:63: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/gazebo_hardware_plugins.dir/all]

2)
--- stderr: dsr_gazebo2
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findgazebo_ros_pkgs.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gazebo_ros_pkgs", but CMake did not find one.

  Could not find a package configuration file provided by "gazebo_ros_pkgs"
  with any of the following names:

    gazebo_ros_pkgsConfig.cmake
    gazebo_ros_pkgs-config.cmake

  Add the installation prefix of "gazebo_ros_pkgs" to CMAKE_PREFIX_PATH or
  set "gazebo_ros_pkgs_DIR" to a directory containing one of the above files.
  If "gazebo_ros_pkgs" provides a separate development package or SDK, be
  sure it has been installed.

3)
--- stderr: dsr_control2 
/usr/bin/ld: cannot find -lPocoFoundation
/usr/bin/ld: cannot find -lPocoNet
/usr/bin/ld: cannot find -lPocoFoundation
/usr/bin/ld: cannot find -lPocoNet
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dsr_control_node2.dir/build.make:330: dsr_control_node2] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/dsr_control_node2.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Failed   <<< dsr_control2 [4.74s, exited with code 2]

doosan-robotics, 2021-04-13:
@whitedoll 

The gazebo_ros_pkgs package is not specified in package.xml.
Install it manually using sudo apt-get install ros-foxy-gazebo-ros-pkgs command, or clone the package again and install it using rosdep command.

whitedoll, 2021-04-18:
i installed foxy-gazebo using command sudo apt-get install ros-foxy-gazebo-ros-pkgs
After running the command source usr/share/gazebo/setup.sh, I reinstalled the Doosan package.
However, the following error is still appearing.

--- stderr: gazebo_ros2_control              
In file included from /home/choi/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:25,
                 from /home/choi/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:19:
/home/choi/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system_interface.hpp:27:10: fatal error: hardware_interface/base_interface.hpp: No such file or directory
   27 | #include "hardware_interface/base_interface.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/gazebo_hardware_plugins.dir/build.make:63: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/gazebo_hardware_plugins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

i would appreciate it if you could fix this error

doosan-robotics, 2021-04-21:
This issue has been fixed.
Please refer to the [link](https://github.com/doosan-robotics/doosan-robot2/issues/5#issuecomment-822975359).

whitedoll, 2021-04-21:
Thanks for solving the "hardware_interface/base_interface.hpp" problem
However, newly build error problem has appeared
error message is here
-----------------------------------------------------------------------------

Starting >>> dsr_control2
[Processing: dsr_control2]                                    
--- stderr: dsr_control2                                       
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:218: warning: "_DEBUG_DSR_CTL" redefined
  218 | #define _DEBUG_DSR_CTL      0
      | 
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_control_node2.cpp:14:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:218: warning: "_DEBUG_DSR_CTL" redefined
  218 | #define _DEBUG_DSR_CTL      0
      | 
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:38: note: this is the location of the previous definition
   38 | #define _DEBUG_DSR_CTL      1
      | 
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_control_node2.cpp:14:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:38: note: this is the location of the previous definition
   38 | #define _DEBUG_DSR_CTL      1
      | 
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:215,
                 from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_control_node2.cpp:14:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/../../../common2/include/DRFL.h:360:6: warning: extra ‘;’ [-Wpedantic]
  360 |     };
      |      ^
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:215,
                 from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/../../../common2/include/DRFL.h:360:6: warning: extra ‘;’ [-Wpedantic]
  360 |     };
      |      ^
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In member function ‘int dsr_control2::DRHWInterface::MsgPublisher_RobotState()’:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:477:60: warning: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  477 |         memcpy(&m_stDrState, &g_stDrState, sizeof(DR_STATE));
      |                                                            ^
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:400:16: note: ‘struct DR_STATE’ declared here
  400 | typedef struct {
      |                ^
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In member function ‘int dsr_control2::DRHWInterface::MsgPublisher_JointState()’:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:620:60: warning: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  620 |         memcpy(&m_stDrState, &g_stDrState, sizeof(DR_STATE));
      |                                                            ^
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:400:16: note: ‘struct DR_STATE’ declared here
  400 | typedef struct {
      |                ^
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:621:26: warning: statement has no effect [-Wunused-value]
  621 |         msg.header.stamp.sec;
      |         ~~~~~~~~~~~~~~~~~^~~
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static void dsr_control2::DRHWInterface::thread_subscribe(rclcpp::Node::SharedPtr)’:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:653:66: warning: unused parameter ‘nh’ [-Wunused-parameter]
  653 |     void DRHWInterface::thread_subscribe(rclcpp::Node::SharedPtr nh)
      |                                          ~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In constructor ‘dsr_control2::DRHWInterface::DRHWInterface(rclcpp::Node::SharedPtr&)’:
/home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:906:52: warning: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
  906 |         memset(&g_stDrState, 0x00, sizeof(DR_STATE));
      |                                                    ^
In file included from /home/choi/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
[... cut ...]
/usr/bin/ld: cannot find -lPocoNet
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dsr_control_node2.dir/build.make:330: dsr_control_node2] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/dsr_control_node2.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< dsr_control2 [36.5s, exited with code 2]

--------------------------------------------------------------------------------------------------------
i would appreciate it if you could fix this error

doosan-robotics, 2021-04-26:
It seems that the poco library is not installed.
Please install the poco library using the command sudo apt-get install libpoco-dev

whitedoll, 2021-04-27:
Thank for response my question
Doosan robot code for ros2 was able to build successfully.

When I ran the example code on the git hub, it worked without any problems.
"ros2 launch dsr_launcher2 single_robot_rviz.launch.py model:=a0912 color:=blue"
"ros2 run dsr_example2_py dsr_service_motion_simple"

However, the Doosan robot that I am using in my project is m1013, and when I ran the following code, I found an error.
"ros2 launch dsr_launcher2 single_robot_rviz.launch.py model:=m1013 color:=white"
"ros2 run dsr_example2_py dsr_service_motion_simple"

ros2 launch dsr_launcher2 single_robot_rviz.launch.py model:=m1013 color:=white
[INFO] [launch]: All log files can be found below /home/choi/.ros/log/2021-04-27-09-33-49-870540-choi-NUC8i7BEH-95701
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sh-1]: process started with pid [95704]
[INFO] [static_transform_publisher-2]: process started with pid [95707]
[INFO] [robot_state_publisher-3]: process started with pid [95709]
[INFO] [rviz2-4]: process started with pid [95712]
[INFO] [dsr_control_node2-5]: process started with pid [95715]
[sh-1] Run Emulator of the Doosan Robot Controller
[sh-1] dirname: /home/choi/ros2_ws2/install/common2/share/common2/bin/DRCF/run_drcf.sh
[sh-1] server_port: 
[sh-1] robot model: 
[sh-1] ARCH: 64-bit
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_0 had 1 children
[robot_state_publisher-3] Link link1 had 1 children
[robot_state_publisher-3] Link link2 had 1 children
[robot_state_publisher-3] Link link3 had 1 children
[robot_state_publisher-3] Link link4 had 1 children
[robot_state_publisher-3] Link link5 had 1 children
[robot_state_publisher-3] Link link6 had 0 children
[robot_state_publisher-3] [INFO] [1619483630.086867869] [robot_state_publisher]: got segment base_0
[robot_state_publisher-3] [INFO] [1619483630.086959994] [robot_state_publisher]: got segment link1
[robot_state_publisher-3] [INFO] [1619483630.086987423] [robot_state_publisher]: got segment link2
[robot_state_publisher-3] [INFO] [1619483630.086992192] [robot_state_publisher]: got segment link3
[robot_state_publisher-3] [INFO] [1619483630.086996028] [robot_state_publisher]: got segment link4
[robot_state_publisher-3] [INFO] [1619483630.087011656] [robot_state_publisher]: got segment link5
[robot_state_publisher-3] [INFO] [1619483630.087017525] [robot_state_publisher]: got segment link6
[robot_state_publisher-3] [INFO] [1619483630.087023553] [robot_state_publisher]: got segment world
[static_transform_publisher-2] [INFO] [1619483630.095059046] [static_transform_publisher]: Spinning until killed publishing transform from 'base' to 'base_0'
[dsr_control_node2-5] [INFO] [1619483630.096009865] [dsr_control_node2]: g_node = 0x0x7ffef0899ed0
[dsr_control_node2-5] [INFO] [1619483630.096093725] [dsr_control_node2]: rate is 100
[dsr_control_node2-5] [INFO] [1619483630.096141440] [dsr_control_node2]: controller_manager is updating!
[dsr_control_node2-5] [INFO] [1619483630.096179113] [dsr_hw_interface2]: name = dsr01
[dsr_control_node2-5] [INFO] [1619483630.096198699] [dsr_hw_interface2]: model = m1013
[dsr_control_node2-5] [INFO] [1619483630.096202187] [dsr_hw_interface2]: gripper = none
[dsr_control_node2-5] [INFO] [1619483630.096205694] [dsr_hw_interface2]: name_space is dsr01, m1013
[dsr_control_node2-5] [INFO] [1619483630.096211343] [dsr_hw_interface2]: constructed
[dsr_control_node2-5] [INFO] [1619483630.131802574] [dsr_hw_interface2]: [dsr_hw_interface2] init() ==> setup callback fucntion
[dsr_control_node2-5] [INFO] [1619483630.131821287] [dsr_hw_interface2]: joint_name = joint1
[dsr_control_node2-5] [WARN] [1619483630.131863815] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[dsr_control_node2-5] [INFO] [1619483630.131921767] [dsr_hw_interface2]: joint_name = joint2
[dsr_control_node2-5] [INFO] [1619483630.131941933] [dsr_hw_interface2]: joint_name = joint3
[dsr_control_node2-5] [INFO] [1619483630.131948215] [dsr_hw_interface2]: joint_name = joint4
[dsr_control_node2-5] [INFO] [1619483630.131957506] [dsr_hw_interface2]: joint_name = joint5
[dsr_control_node2-5] [INFO] [1619483630.131963294] [dsr_hw_interface2]: joint_name = joint6
[dsr_control_node2-5] [INFO] [1619483630.131967791] [dsr_hw_interface2]: INIT@@@@@@@@@@@@@@@@@@@@@@@@@
[dsr_control_node2-5] [INFO] [1619483630.131971568] [dsr_hw_interface2]: init() ==> arm is standby
[dsr_control_node2-5] [INFO] [1619483630.131980265] [dsr_hw_interface2]: host = 127.0.0.1
[dsr_control_node2-5] [INFO] [1619483630.131983246] [dsr_hw_interface2]: port = 12345
[dsr_control_node2-5] [INFO] [1619483630.131986115] [dsr_hw_interface2]: command = 1
[dsr_control_node2-5] [INFO] [1619483630.131989143] [dsr_hw_interface2]: mode = virtual
[dsr_control_node2-5] [INFO] [1619483630.131991986] [dsr_hw_interface2]: host 127.0.0.1, port=12345 bCommand: 1, mode: virtual
[dsr_control_node2-5] [INFO] [1619483630.132349519] [dsr_hw_interface2]: On Monitor State
[dsr_control_node2-5] [INFO] [1619483630.132478512] [dsr_hw_interface2]: On Monitor State
[... cut ...]
[dsr_control_node2-5] [INFO] [1619483630.152291558] [dsr_control_node2]: controller_manager is updating!
[rviz2-4] [INFO] [1619483630.432382361] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1619483630.432587390] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1619483630.456505788] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] Parsing robot urdf xml string.
[rviz2-4] Parsing robot urdf xml string.
[rviz2-4] Parsing robot urdf xml string.

movie link : https://drive.google.com/file/d/19MpCc0w617BXZVhMytN-y3grVTWUuCjj/view?usp=sharing

doosan-robotics, 2021-04-27:
@whitedoll 

The same problem occurred in another issue and that part was corrected.
This was caused by not installing the libpoco-dev package.
Added description of package installation to Readme.md.
Install the poco library for the connection using the command sudo apt-get install libpoco-dev.

closed 2022-08-31 by doosan-robotics