DoosanRobotics/doosan-robot2 #5: [ERROR] build problem in gazebo_ros2_control

bleut, 2021-04-15:
Hello, I'm using Ubuntu 20.04 and I installed ros2 foxy via Debian package
I got an error in installing doosan-robot2 package.

I got an error saying 

<code>
--- stderr: gazebo_ros2_control
In file included from /home/hmdo/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system.hpp:25,
                 from /home/hmdo/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/src/gazebo_system.cpp:19:
/home/hmdo/ros2_ws/src/gazebo_ros2_control/gazebo_ros2_control/include/gazebo_ros2_control/gazebo_system_interface.hpp:27:10: fatal error: hardware_interface/base_interface.hpp: No such file or directory
   27 | #include "hardware_interface/base_interface.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/gazebo_hardware_plugins.dir/build.make:63: CMakeFiles/gazebo_hardware_plugins.dir/src/gazebo_system.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/gazebo_hardware_plugins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
</code>

Pls help me to fix it.

doosan-robotics, 2021-04-20:
Our package is not compatible with the latest gazebo_ros2_control package. Use the previous version of gazebo_ros2_control with the command below.

$ cd gazebo_ros_control && git reset --hard 3dfe04d412d5be4540752e9c1165ccf25d7c51fb && cd ..

The contents have been updated to Readme.md.

bleut, 2021-04-27:
I think this issue was fixed. Thank you for your help.

closed 2021-04-27 by bleut

Sorawit-DeepX, 2021-07-08:
I still found this problem using
cd gazebo_ros2_control && git reset --hard 3dfe04d412d5be4540752e9c1165ccf25d7c51fb && cd ..

Have to delete dsr2_control to make dsr2_description work.
Any advise?