DoosanRobotics/doosan-robot2 #8: dsr_control2 build fail

frvd, 2021-05-08:
I followed the steps described in the readme, but I still get this error when trying to compile.

Starting >>> dsr_control2
--- stderr: dsr_control2                                                                                                                                                                                        
In file included from ~/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:9:
~/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:54:14: fatal error: hardware_interface/joint_handle.hpp: No such file or directory
   54 |     #include "hardware_interface/joint_handle.hpp"
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from ~/src/doosan-robot2/dsr_control2/src/dsr_control_node2.cpp:14:
~/src/doosan-robot2/dsr_control2/include/dsr_control2/dsr_hw_interface2.h:54:14: fatal error: hardware_interface/joint_handle.hpp: No such file or directory
   54 |     #include "hardware_interface/joint_handle.hpp"
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/dsr_control_node2.dir/build.make:63: CMakeFiles/dsr_control_node2.dir/src/dsr_hw_interface2.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/dsr_control_node2.dir/build.make:76: CMakeFiles/dsr_control_node2.dir/src/dsr_control_node2.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/dsr_control_node2.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

doosan-robotics, 2021-05-09:
Does the same error occur even if I refer to the process in README.md?
The doosan-robot2 package is not yet compatible with the latest ros2_control, ros2_controller, and gazebo_ros2_control packages. As stated in README.md, you must load the previous version of the commit using the git reset command.

frvd, 2021-05-09:
It does. I am relying on the fact that ROS2 is working like ROS1, in the sense that the packages in my development workspace are prioritized over the system-wide packages. Maybe this is not the case and I have to uninstall ros2_control & co ?

dvalenciar, 2021-08-23:
Same problem here with: 
   54 |     #include "hardware_interface/joint_handle.hpp"

is the package update to work with the last version of ROS2_CONTROL?

frvd, 2021-08-24:
@dvalenciar No, it is not updated with the last version. The readme states to get the old ros2_control versions in your workspace to compile against it, but even if I do that it doesn't compile for me

closed 2025-01-13 by doosan-robotics