DoosanRobotics/doosan-robot2 #9: The robot stops moving while the function set_desired_force() is running.

whitedoll, 2021-05-17:
Hi
I am using m1013 model in my project and after connecting ros2 in real mode, movel or movej command works well without error.
However, when moving the robot using the set_desire_force command, I found an error that stops motion within 2 seconds.

I don't know how to debug when moving the robot with ros2 service, sorry
I would be grateful if I could fix the problem

doosan-robotics, 2021-05-26:
Currently, a problem was found with the set_desired_force function inside the API.
It will be revised in the near future.

focusonfocus, 2021-09-13:
Having a problem with set_desired_force, is it updated?

doosan-robotics, 2023-05-08:
I'm sorry for the late update. Now, API about force control is updated in order to fix a problem with set_desired_force.

closed 2023-05-25 by doosan-robotics