DoosanRobotics/doosan-robot2 #10: Problem running dsr_control_node2

JTaveau, 2021-06-22:
Hi, 
Every time I try to run the dsr_control_node2 with the launch files, I get this error :

I used ros2 launch dsr_control2 dsr_control2.launch.py, instead of the rviz launch to have only this error.  

[INFO] [launch]: All log files can be found below /home/jtaveau/.ros/log/2021-06-22-11-03-49-470003-saluyt-247231
[INFO] [launch]: Default logging verbosity is set to INFO
/home/jtaveau/moveit2_ws/install/dsr_control2/share/dsr_control2/config/default.yaml
[INFO] [dsr_control_node2-1]: process started with pid [247240]
[dsr_control_node2-1] [INFO] [1624352629.578406090] [dsr_control_node2]: g_node = 0x0x7fff587a7410
[dsr_control_node2-1] [INFO] [1624352629.578591177] [dsr_control_node2]: rate is 100
[dsr_control_node2-1] [INFO] [1624352629.578703692] [dsr_control_node2]: controller_manager is updating!
[dsr_control_node2-1] [INFO] [1624352629.578949447] [dsr_hw_interface2]: name = dsr01
[dsr_control_node2-1] [INFO] [1624352629.578970939] [dsr_hw_interface2]: model = m1013
[dsr_control_node2-1] [INFO] [1624352629.578976680] [dsr_hw_interface2]: gripper = none
[dsr_control_node2-1] [INFO] [1624352629.578982193] [dsr_hw_interface2]: name_space is dsr01, m1013
[dsr_control_node2-1] [INFO] [1624352629.578986618] [dsr_hw_interface2]: constructed
[dsr_control_node2-1] [INFO] [1624352629.648201131] [dsr_hw_interface2]: [dsr_hw_interface2] init() ==> setup callback fucntion
[dsr_control_node2-1] [INFO] [1624352629.648250886] [dsr_hw_interface2]: joint_name = joint1
[dsr_control_node2-1] [INFO] [1624352629.648379565] [dsr_hw_interface2]: joint_name = joint2
[dsr_control_node2-1] [INFO] [1624352629.648409293] [dsr_hw_interface2]: joint_name = joint3
[dsr_control_node2-1] [INFO] [1624352629.648426622] [dsr_hw_interface2]: joint_name = joint4
[dsr_control_node2-1] [INFO] [1624352629.649017060] [dsr_hw_interface2]: joint_name = joint5
[dsr_control_node2-1] [INFO] [1624352629.649036958] [dsr_hw_interface2]: joint_name = joint6
[dsr_control_node2-1] [INFO] [1624352629.649049081] [dsr_hw_interface2]: INIT@@@@@@@@@@@@@@@@@@@@@@@@@
[dsr_control_node2-1] [INFO] [1624352629.649058139] [dsr_hw_interface2]: init() ==> arm is standby
[dsr_control_node2-1] [INFO] [1624352629.649122038] [dsr_hw_interface2]: host = 127.0.0.1
[dsr_control_node2-1] [INFO] [1624352629.649129870] [dsr_hw_interface2]: port = 12345
[dsr_control_node2-1] [INFO] [1624352629.649134501] [dsr_hw_interface2]: command = 1
[dsr_control_node2-1] [INFO] [1624352629.649138682] [dsr_hw_interface2]: mode = virtual
[dsr_control_node2-1] [INFO] [1624352629.649142855] [dsr_hw_interface2]: host 127.0.0.1, port=12345 bCommand: 1, mode: virtual
[dsr_control_node2-1] joint_name = joint1joint_name = joint2joint_name = joint3joint_name = joint4joint_name = joint5joint_name = joint6Connection refused
[dsr_control_node2-1] [ERROR] [1624352629.649930486] [dsr_hw_interface2]: DRHWInterface::init() DRCF connecting ERROR!!!
[dsr_control_node2-1] [ERROR] [1624352629.649953211] [dsr_control_node2]: Error initializing robot
[dsr_control_node2-1] [INFO] [1624352629.649986810] [dsr_hw_interface2]: DRHWInterface::~DRHWInterface()
[ERROR] [dsr_control_node2-1]: process has died [pid 247240, exit code -11, cmd '/home/jtaveau/moveit2_ws/install/dsr_control2/lib/dsr_control2/dsr_control_node2 --ros-args --params-file /tmp/launch_params_sw2nerns --params-file /tmp/launch_params_d_fnv78q --params-file /tmp/launch_params_tzx9c1k8 --params-file /tmp/launch_params_5imt5lon --params-file /tmp/launch_params_n9gdnyt7 --params-file /tmp/launch_params_5f_3wwyg --params-file /tmp/launch_params_q4vvj7g3 --params-file /tmp/launch_params_0twxt0ha --params-file /tmp/launch_params_b9kz3rvs --params-file /tmp/launch_params_v0x1_gxz'].

Could you help me, please ? 
I'm using Foxy on Debian Bullseye.

kong2983, 2021-06-24:
Did the emulator run normally?
Try running the DRCF64 DRCF binary file in common2/bin/DRCF.
If it runs normally, install the lib-pocodev package and run it again.

JTaveau, 2021-06-24:
When I run it, I get : 
(env38) jtaveau@saluyt:~/doosan_ws$ ./src/doosan-robot2/common2/bin/DRCF/DRCF64 
_______________________________________________________________
        Emulator of Doosan Robot Controller [ver M2.52]               
                  [service port = 12345]                          
                  [robot model = m1013(config/config.xml)]                       
_______________________________________________________________
Does it mean it runs normally ? 
I have already installed libpoco-dev.

closed 2021-06-25 by JTaveau