DoosanRobotics/doosan-robot2 #14: Client T/P authentication failed.

Sorawit-DeepX, 2021-08-18:
Hi guys.

I've successfully built `doosan-robot2` and now I'm trying to use it with a real m0609 robot.

I use command `ros2 launch dsr_launcher2 single_robot_rviz.launch.py mode:=real host:=192.168.137.100 port:=12345 model:=m0609`

Command output
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2021-08-18-00-03-50-786022-3fa24d1bd2a7-43057
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [static_transform_publisher-1]: process started with pid [43060]
[INFO] [robot_state_publisher-2]: process started with pid [43062]
[INFO] [rviz2-3]: process started with pid [43064]
[INFO] [dsr_control_node2-4]: process started with pid [43066]
[static_transform_publisher-1] [INFO] [1629245031.019471688] [static_transform_publisher]: Spinning until killed publishing transform from 'base' to 'base_0'
[dsr_control_node2-4] [INFO] [1629245031.034142435] [dsr_control_node2]: g_node = 0x0x7ffff160b940
[dsr_control_node2-4] [INFO] [1629245031.034239149] [dsr_control_node2]: rate is 100
[dsr_control_node2-4] [INFO] [1629245031.034293015] [dsr_control_node2]: controller_manager is updating!
[dsr_control_node2-4] [INFO] [1629245031.034437427] [dsr_hw_interface2]: name = dsr01
[dsr_control_node2-4] [INFO] [1629245031.034488501] [dsr_hw_interface2]: model = m0609
[dsr_control_node2-4] [INFO] [1629245031.034506933] [dsr_hw_interface2]: gripper = none
[dsr_control_node2-4] [INFO] [1629245031.034538973] [dsr_hw_interface2]: name_space is dsr01, m0609
[dsr_control_node2-4] [INFO] [1629245031.034555807] [dsr_hw_interface2]: constructed
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link base_0 had 1 children
[robot_state_publisher-2] Link link1 had 1 children
[robot_state_publisher-2] Link link2 had 1 children
[robot_state_publisher-2] Link link3 had 1 children
[robot_state_publisher-2] Link link4 had 1 children
[robot_state_publisher-2] Link link5 had 1 children
[robot_state_publisher-2] Link link6 had 0 children
[robot_state_publisher-2] [INFO] [1629245031.037395410] [robot_state_publisher]: got segment base_0
[robot_state_publisher-2] [INFO] [1629245031.037522875] [robot_state_publisher]: got segment link1
[robot_state_publisher-2] [INFO] [1629245031.037546068] [robot_state_publisher]: got segment link2
[robot_state_publisher-2] [INFO] [1629245031.037560954] [robot_state_publisher]: got segment link3
[robot_state_publisher-2] [INFO] [1629245031.037574242] [robot_state_publisher]: got segment link4
[robot_state_publisher-2] [INFO] [1629245031.037587512] [robot_state_publisher]: got segment link5
[robot_state_publisher-2] [INFO] [1629245031.037601042] [robot_state_publisher]: got segment link6
[robot_state_publisher-2] [INFO] [1629245031.037613645] [robot_state_publisher]: got segment world
[rviz2-3] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[dsr_control_node2-4] [INFO] [1629245031.080160541] [dsr_hw_interface2]: [dsr_hw_interface2] init() ==> setup callback fucntion
[dsr_control_node2-4] [INFO] [1629245031.080194003] [dsr_hw_interface2]: joint_name = joint1
[dsr_control_node2-4] [WARN] [1629245031.080242148] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
```

However Rviz give multiple warning.
```
[rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "link2" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "link3" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "link4" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "link5" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "link6" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
```
at the same time robot's teach pendant displays the following message:
`Client T/P authentication failed.`
Cannot find any related topic with the error displayed.

Any advice would be most welcome!

mpavezb, 2021-08-24:
Hi @Sorawit-DeepX, I'm linking similar issue for ROS1: https://github.com/doosan-robotics/doosan-robot2/issues/14

doosan-robotics, 2023-03-02:
@ghost
Are you trying to use it in docker?
When used in docker, TP authentication may fail because the virtual port number and the actual port number do not match. Please try again after matching the port number on docker with the actual physical port number.

closed 2023-03-08 by doosan-robotics