DoosanRobotics/doosan-robot2 #16: hardware_interface fail Windows

ClemGRob, 2021-11-22:
hello, i'm trying to use your package with ros2 foxy on Windows 10, and i'm having an issue when I try to use colcon.

this is what I do : 
on a regular prompt
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws/src
git clone https://github.com/doosan-robotics/doosan-robot2.git
git clone -b foxy https://github.com/ros-controls/ros2_control.git
git clone -b foxy https://github.com/ros-controls/ros2_controllers.git
git clone -b foxy https://github.com/ros-simulation/gazebo_ros2_control.git
cd ros2_control && git reset --hard 3dc62e28e3bc8cf636275825526c11d13b554bb6 && cd ..
cd ros2_controllers && git reset --hard 83c494f460f1c8675f4fdd6fb8707b87e81cb197 && cd ..
cd gazebo_ros2_control && git reset --hard 3dfe04d412d5be4540752e9c1165ccf25d7c51fb && cd ..
git clone -b ros2 --single-branch https://github.com/ros-planning/moveit_msgs
cd ~/ros2_ws
rosdep update
rosdep install --from-paths src --ignore-src --rosdistro foxy -r -y

and on a x64 prompt
colcon build

this is my output

[1.258s] root DEBUG Using proactor: IocpProactor
Starting >>> hardware_interface
Starting >>> controller_manager_msgs                        
Starting >>> dsr_msgs2                                                                                          
Starting >>> moveit_msgs
[Processing: controller_manager_msgs, dsr_msgs2, hardware_interface, moveit_msgs]
Failed   <<< hardware_interface [43.3s, exited with code 1]
Aborted  <<< controller_manager_msgs [1min 0s]
Aborted  <<< moveit_msgs [1min 23s]
Aborted  <<< dsr_msgs2 [1min 24s]
Summary: 0 packages finished [1min 25s]
1 package failed: hardware_interface
3 packages aborted: controller_manager_msgs dsr_msgs2 moveit_msgs
2 packages had stderr output: dsr_msgs2 moveit_msgs
22 packages not processed

do you know where the problem comes from, and how to solve it?

doosan-robotics, 2021-12-06:
Currently not tested on Windows.
Are you using ROS2 on Windows?

doosan-robotics, 2023-02-23:
Can you try on WSL environment? It was confirmed that it operates normally in a WSL environment.

Here is the link you can follow [https://keep-steady.tistory.com/45](url).

closed 2023-03-08 by doosan-robotics