DoosanRobotics/doosan-robot2 #22: example error

junghs1040, 2022-07-12:
Hi i got an error when i tried to run the examples
can you help me?

$ ros2 run dsr_example2_py dsr_service_motion_simple_class
_robot_id =
_robot_model =
_srv_name_prefix =
_topic_name_prefix =
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'dsr_msgs2.dsr_msgs2_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/seok2/ros2_ws/install/dsr_example2_py/lib/dsr_example2_py/dsr_service_motion_simple_class", line 11, in <module>
    load_entry_point('dsr-example2-py==0.0.0', 'console_scripts', 'dsr_service_motion_simple_class')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/seok2/ros2_ws/install/dsr_example2_py/lib/python3.8/site-packages/dsr_example2_py/dsr_service_motion_simple_class.py", line 24, in <module>
    from DSR_ROBOT2 import *
  File "/home/seok2/ros2_ws/install/common2/bin/common2/imp/DSR_ROBOT2.py", line 52, in <module>
    _ros2_set_robot_mode             = g_node.create_client(SetRobotMode,           _srv_name_prefix +"system/set_robot_mode")
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/node.py", line 1249, in create_client
    check_for_type_support(srv_type)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_type.__class__.__import_type_support__()
  File "/home/seok2/ros2_ws/install/dsr_msgs2/lib/python3.9/site-packages/dsr_msgs2/srv/_set_robot_mode.py", line 256, in __import_type_support__
    module = import_type_support('dsr_msgs2')
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'dsr_msgs2'

jerry-loungelab, 2022-07-13:
I have same problem.
~/doosan_ws$ ros2 run dsr_example2_py dsr_service_motion_simple
_robot_id =
_robot_model =
_srv_name_prefix =
_topic_name_prefix =
Traceback (most recent call last):
  File "/home/loungelab/doosan_ws/install/dsr_example2_py/lib/dsr_example2_py/dsr_service_motion_simple", line 11, in <module>
    load_entry_point('dsr-example2-py==0.0.0', 'console_scripts', 'dsr_service_motion_simple')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/loungelab/doosan_ws/build/dsr_example2_py/dsr_example2_py/dsr_service_motion_simple.py", line 25, in <module>
    from DSR_ROBOT2 import *
  File "/home/loungelab/doosan_ws/src/doosan-robot2/common2/imp/DSR_ROBOT2.py", line 52, in <module>
    _ros2_set_robot_mode             = g_node.create_client(SetRobotMode,           _srv_name_prefix +"system/set_robot_mode")
AttributeError: 'str' object has no attribute 'create_client'

doosan-robotics, 2023-02-21:
Didn't you go through a separate modification?

closed 2023-03-08 by doosan-robotics