DoosanRobotics/doosan-robot2 #25: [ERROR] moveit example build error

Rhymakers, 2022-08-11:
colcon build error occured while building moveit example.
build result below.

code:
$ colcon build
[0.332s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'controller_interface' is in: /home/yh99/ros2_ws/install/controller_interface, /opt/ros/foxy
	'controller_manager' is in: /home/yh99/ros2_ws/install/controller_manager
	'hardware_interface' is in: /home/yh99/ros2_ws/install/hardware_interface, /opt/ros/foxy
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding controller_interface controller_manager hardware_interface

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> geometric_shapes
Starting >>> moveit_common
Starting >>> warehouse_ros
Starting >>> hardware_interface
Starting >>> controller_manager_msgs
Starting >>> dsr_description2
Starting >>> moveit_resources_prbt_support
Starting >>> dsr_msgs2
Starting >>> common2
Starting >>> dsr_launcher2
Finished <<< moveit_common [0.56s]                                           
Finished <<< common2 [0.54s]                                          
Finished <<< moveit_resources_prbt_support [0.56s]
Starting >>> turtlesim
Finished <<< dsr_description2 [0.63s]
Finished <<< dsr_launcher2 [0.63s]                                    
Finished <<< warehouse_ros [0.70s]
Starting >>> warehouse_ros_mongo
Finished <<< srdfdom [0.77s]
Starting >>> moveit_configs_utils
Finished <<< hardware_interface [0.80s]                               
Finished <<< geometric_shapes [0.83s]
Starting >>> controller_interface
Starting >>> test_robot_hardware
Starting >>> joint_limits_interface
Starting >>> transmission_interface
Finished <<< controller_manager_msgs [1.11s]                           
Finished <<< controller_interface [0.50s]                              
Finished <<< warehouse_ros_mongo [0.68s]                               
Finished <<< joint_limits_interface [0.59s]
Finished <<< test_robot_hardware [0.61s]
Starting >>> controller_manager
Finished <<< transmission_interface [0.62s]                            
Finished <<< turtlesim [1.08s]                                         
Finished <<< controller_manager [0.32s]
Starting >>> forward_command_controller
Starting >>> joint_trajectory_controller                               
Starting >>> joint_state_controller
Starting >>> diff_drive_controller
Starting >>> gazebo_ros2_control
Starting >>> ros2_control
Finished <<< moveit_configs_utils [1.09s]
Finished <<< ros2_control [0.23s]                                      
Finished <<< gazebo_ros2_control [0.35s]                               
Finished <<< forward_command_controller [0.40s]
Starting >>> position_controllers                                      
Starting >>> velocity_controllers
Starting >>> effort_controllers
Finished <<< joint_state_controller [0.45s]
Finished <<< diff_drive_controller [0.59s]                          
Finished <<< joint_trajectory_controller [0.67s]
Starting >>> ros2_controllers
Finished <<< effort_controllers [0.40s]                             
Finished <<< position_controllers [0.44s]
Finished <<< velocity_controllers [0.44s]
Starting >>> dsr_gazebo2
Starting >>> gazebo_ros2_control_demos
Finished <<< ros2_controllers [0.22s]
Starting >>> fake_joint_driver
Finished <<< dsr_gazebo2 [0.19s]                                    
Finished <<< gazebo_ros2_control_demos [0.21s]
Finished <<< fake_joint_driver [0.23s]                              
Finished <<< moveit_msgs [4.04s]                                    
Starting >>> moveit_core
--- stderr: moveit_core                                           
/home/yh99/ros2_ws/src/moveit2/moveit_core/transforms/src/transforms.cpp:41:10: fatal error: tf2_eigen/tf2_eigen.hpp: No such file or directory
   41 | #include <tf2_eigen/tf2_eigen.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [transforms/CMakeFiles/moveit_transforms.dir/build.make:63: transforms/CMakeFiles/moveit_transforms.dir/src/transforms.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2143: transforms/CMakeFiles/moveit_transforms.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::getCurrentlyLoadingLibraryName[abi:cxx11]()'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::getPluginBaseToFactoryMapMapMutex()'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::hasANonPurePluginLibraryBeenOpened(bool)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::getCurrentlyActiveClassLoader()'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::AbstractMetaObjectBase::setAssociatedLibraryPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::getFactoryMapForBaseClass(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::AbstractMetaObjectBase::AbstractMetaObjectBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.5.3: undefined reference to `class_loader::impl::AbstractMetaObjectBase::addOwningClassLoader(class_loader::ClassLoader*)'
collect2: error: ld returned 1 exit status
make[2]: *** [online_signal_smoothing/CMakeFiles/test_butterworth_filter.dir/build.make:216: online_signal_smoothing/test_butterworth_filter] Error 1
make[1]: *** [CMakeFiles/Makefile2:1330: online_signal_smoothing/CMakeFiles/test_butterworth_filter.dir/all] Error 2
/home/yh99/ros2_ws/src/moveit2/moveit_core/distance_field/src/distance_field.cpp:43:10: fatal error: tf2_eigen/tf2_eigen.hpp: No such file or directory
   43 | #include <tf2_eigen/tf2_eigen.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [distance_field/CMakeFiles/moveit_distance_field.dir/build.make:63: distance_field/CMakeFiles/moveit_distance_field.dir/src/distance_field.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1071: distance_field/CMakeFiles/moveit_distance_field.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_core [0.54s, exited with code 2]
Aborted  <<< dsr_msgs2 [6.03s]                               

Summary: 32 packages finished [6.45s]
  1 package failed: moveit_core
  1 package aborted: dsr_msgs2
  1 package had stderr output: moveit_core
  44 packages not processed

benthebear93, 2022-08-16:
First Check if moveit2 is installd correctly by running below command

`ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true`

Rhymakers, 2022-08-16:
**This issue has been resolved.**
And since moveit_tutorial is not in the given installation process does it required for this package to run?

**Thank you for your answer!!**

soultion

```bash
$ cd ~/ros2_ws/src
$ git clone https://github.com/ros-planning/moveit2
```
After getting moveit2 package user have to run ``git checkout``, ``git switch`` to ensure getting right branch of moveit2.
```bash
$ cd ~/ros2_ws/src/moveit2
$ git checkout foxy
$ git switch
```

benthebear93, 2022-08-16:
@Rhymakers
Nope, you don't need a moveit_tutorial to run this pkg 

Just recommd to test moveit2, since it is the basic tutorial given by [moveit2 ](https://moveit.picknik.ai/foxy/doc/getting_started/getting_started.html)  :)

closed 2022-08-16 by Rhymakers