DoosanRobotics/doosan-robot2 #27: Is this abandoned?

jaanush, 2022-11-23:
Is this abandonware? Is it even meaningful to use?

sampreets3, 2022-11-23:
Hey @jaanush, I think it might be the case. I asked about updates on when they might port to humble in another issue, but nobody from Doosan responded. With Foxy coming to EOL in 2023, they should really think about releasing updates for Humble. If not, we might have to hack our way in to at least be using the robot.

doosan-robotics, 2023-02-20:
First of all, sorry for the delay in answering. We recognize that the support period for the foxy version is until 2023 and will soon update the current Doosan robot ros2 foxy version to match the latest package (e.g. ros2_control, ros2_controller, gazebo_ros2_control). The humble version will proceed after the previous work. Many updates will be made in the future, so please pay a lot of attention.

closed 2023-02-22 by doosan-robotics