DoosanRobotics/doosan-robot2 #33: wsl 상에서 패키지 구동시 오류

hanprd, 2023-05-15:
안녕하세요.

현재 windows 11에서 wsl2 (ubuntu 20.04.6 LTS)를 활용하여 doosan-robot2 패키지를 구동하려 시도하고 있습니다.
ROS2 foxy에 doosan-robot2 패키지를 설치하였고 그 과정에서 아래와 같은 메시지를 보았지만 무시 가능한 경고 메시지라는 이야기를 들었습니다.

User
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::get_external_torque_cb(std::shared_ptr<dsr_msgs2::srv::GetExternalTorque_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetExternalTorque_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:1997:114: warning: unused parameter ‘req’ [-Wunused-parameter]
 1997 | WInterface::get_external_torque_cb(const std::shared_ptr<dsr_msgs2::srv::GetExternalTorque::Request> req, std::shared_ptr<dsr_msgs2::srv::GetExternalTorque::Response> res)
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::get_tool_force_cb(std::shared_ptr<dsr_msgs2::srv::GetToolForce_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetToolForce_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2010:104: warning: unused parameter ‘req’ [-Wunused-parameter]
 2010 |   bool DRHWInterface::get_tool_force_cb(const std::shared_ptr<dsr_msgs2::srv::GetToolForce::Request> req, std::shared_ptr<dsr_msgs2::srv::GetToolForce::Response> res)
      |                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::get_workpiece_weight_cb(std::shared_ptr<dsr_msgs2::srv::GetWorkpieceWeight_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetWorkpieceWeight_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2059:116: warning: unused parameter ‘req’ [-Wunused-parameter]
 2059 | nterface::get_workpiece_weight_cb(const std::shared_ptr<dsr_msgs2::srv::GetWorkpieceWeight::Request> req, std::shared_ptr<dsr_msgs2::srv::GetWorkpieceWeight::Response> res)
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::reset_workpiece_weight_cb(std::shared_ptr<dsr_msgs2::srv::ResetWorkpieceWeight_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::ResetWorkpieceWeight_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2069:120: warning: unused parameter ‘req’ [-Wunused-parameter]
 2069 | face::reset_workpiece_weight_cb(const std::shared_ptr<dsr_msgs2::srv::ResetWorkpieceWeight::Request> req, std::shared_ptr<dsr_msgs2::srv::ResetWorkpieceWeight::Response> res)
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::release_compliance_ctrl_cb(std::shared_ptr<dsr_msgs2::srv::ReleaseComplianceCtrl_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::ReleaseComplianceCtrl_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2182:122: warning: unused parameter ‘req’ [-Wunused-parameter]
 2182 | ce::release_compliance_ctrl_cb(const std::shared_ptr<dsr_msgs2::srv::ReleaseComplianceCtrl::Request> req, std::shared_ptr<dsr_msgs2::srv::ReleaseComplianceCtrl::Response> res)
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::drl_pause_cb(std::shared_ptr<dsr_msgs2::srv::DrlPause_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::DrlPause_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2746:95: warning: unused parameter ‘req’ [-Wunused-parameter]
 2746 |     bool DRHWInterface::drl_pause_cb(const std::shared_ptr<dsr_msgs2::srv::DrlPause::Request> req, std::shared_ptr<dsr_msgs2::srv::DrlPause::Response> res)
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::drl_resume_cb(std::shared_ptr<dsr_msgs2::srv::DrlResume_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::DrlResume_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2782:97: warning: unused parameter ‘req’ [-Wunused-parameter]
 2782 |     bool DRHWInterface::drl_resume_cb(const std::shared_ptr<dsr_msgs2::srv::DrlResume::Request> req, std::shared_ptr<dsr_msgs2::srv::DrlResume::Response> res)
      |                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::get_drl_state_cb(std::shared_ptr<dsr_msgs2::srv::GetDrlState_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetDrlState_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2794:102: warning: unused parameter ‘req’ [-Wunused-parameter]
 2794 |     bool DRHWInterface::get_drl_state_cb(const std::shared_ptr<dsr_msgs2::srv::GetDrlState::Request> req, std::shared_ptr<dsr_msgs2::srv::GetDrlState::Response> res)
      |                                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::get_current_tcp_cb(std::shared_ptr<dsr_msgs2::srv::GetCurrentTcp_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetCurrentTcp_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2817:106: warning: unused parameter ‘req’ [-Wunused-parameter]
 2817 | bool DRHWInterface::get_current_tcp_cb(const std::shared_ptr<dsr_msgs2::srv::GetCurrentTcp::Request> req, std::shared_ptr<dsr_msgs2::srv::GetCurrentTcp::Response> res)
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::get_current_tool_cb(std::shared_ptr<dsr_msgs2::srv::GetCurrentTool_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::GetCurrentTool_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2850:108: warning: unused parameter ‘req’ [-Wunused-parameter]
 2850 | ol DRHWInterface::get_current_tool_cb(const std::shared_ptr<dsr_msgs2::srv::GetCurrentTool::Request> req, std::shared_ptr<dsr_msgs2::srv::GetCurrentTool::Response> res)
      |                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~

/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp: In static member function ‘static bool dsr_control2::DRHWInterface::robotiq_2f_open_cb(std::shared_ptr<dsr_msgs2::srv::Robotiq2FOpen_Request_<std::allocator<void> > >, std::shared_ptr<dsr_msgs2::srv::Robotiq2FOpen_Response_<std::allocator<void> > >)’:
/home/han/ros2_ws/src/doosan-robot2/dsr_control2/src/dsr_hw_interface2.cpp:2915:106: warning: unused parameter ‘req’ [-Wunused-parameter]
 2915 | bool DRHWInterface::robotiq_2f_open_cb(const std::shared_ptr<dsr_msgs2::srv::Robotiq2FOpen::Request> req, std::shared_ptr<dsr_msgs2::srv::Robotiq2FOpen::Response> res)
[... cut ...] ansition 1 from current state inactive: Transition is not registered., at /tmp/binarydeb/ros-foxy-rcl-lifecycle-1.1.14/src/rcl_lifecycle.c:350   [gzserver-4] [ERROR] [1684165518.292223890] [gazebo_ros2_control]: Loaded gazebo_ros2_control.**                 

3. Real 모드로 연결시 [1.3004] Client T/P authenticiation failed. 메시지가 T/P 상에 출력됩니다. 두산 로봇 github 에서 비슷한 사례를 보았고 포트 포워딩을 해보았지만 문제가 해결되지 않습니다. Ping 명령어로 통신을 확인했을 때는 이상없이 통신이 됩니다. 아예 방화벽을 꺼보기도 했지만 문제가 해결되지 않네요.

위 이슈에 대한 해결 방안을 좀 자세히 알려주시면 감사하겠습니다.

doosan-robotics, 2024-01-24:
안녕하세요 

상세한 상황 설명 감사합니다. 적어주신 번호 순대로 답변드리겠습니다.
1. foxy 패키지에 dependency 업데이트가 되면서 다음과 같은 현상이 발생하는 것을 발견하였습니다. 내부적으로 인지하고 있으며 업데이트 되는 대로 안내 드리겠습니다  해당 현상은 rviz를 단독으로 사용할 경우에만 발생하는 이슈로 gazebo+rviz를 사용하면 가능합니다. 
2. 해당 메세지는 gazebo 런치파일을 실행하면서 발생하는 로그를  /home/han/.ros/log/2023-05-15-11-45-14-261617-han-hp-1474  에 저장한다는 의미입니다. 
3. ROS 패키지를 이용하여 실제로봇 제어 할 경우에 TP사용을 동시 사용 하지 못하도록 제어권 정책이 되어져있습니다. (동시다발 제어로 위험사항을 막기위함)

감사합니다.

creep-ant, 2024-03-04:
안녕하세요 @doosan-robotics 님.

1번 과 같은 내용에서 저도 문제를 겪고 있는데요.
rviz를 단독으로 사용하지 않고 gazebo + rviz를 사용한다는 내용에 대한 설명을 부탁드립니다.

감사합니다.

doosan-robotics, 2025-02-19:
안녕하세요 

해당이슈는 humble 버전 으로 대체하며 해결이 가능 한 것으로 보여 해당 이슈는 종결 하겠습니다 
추가적인 이슈가 발생하면 새 이슈로 등록 부탁드립니다.