DoosanRobotics/doosan-robot2 #35: gazebo_ros2_control package colcon build error

DevelopersNotes, 2023-07-07:
ROS2 Foxy EOL 이후 새로운 우분투20.04 환경에 Foxy를 설치하고 본 소스를 다시 빌드 해보려는데 다음과 같은 오류가 발생하여 해결을 못하고 있습니다.

CMake Error in CMakeLists.txt:
  Imported target "dart" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< gazebo_ros2_control

혹시 해결 방법이 있는지 이슈를 올려봅니다.

Lau-JW, 2024-09-11:
you can open this cmakelist file, and annotate line 19; Link this "#find_package(gazebo_ros_pkgs REQUIRED)", gazebo_ros_pkgs is not suitable for new version gazebo.

doosan-robotics, 2025-02-19:
This issue can be resolved with the Humble version,
 it will be closed accordingly.

closed 2025-02-19 by doosan-robotics