DoosanRobotics/doosan-robot2 #40: ROS2-HUMBLE upgrade

huand, 2023-09-26:
Ros2-Foxy has reached End Of Life.

There are no clear open issues on this matter.

Related closed issues:  #27 .

Citing @doosan-robotics on Feb 20 (7 months ago)

> Many updates will be made in the future, so please pay a lot of attention.

I think we can use this ticket to track the progress, timeline, etc. As long as the migration is not done, it's still an open issue for us who wants to integrate Doosan Robots into up-to-date ROS2 systems.

bmdyrdal, 2024-01-10:
Any update to this? @doosan-robotics

song-ms, 2024-01-17:
@bmdyrdal @huand 
We aim to release humble on 24' May. 
if we have any update, I will inform you

doosan-robotics, 2025-02-19:
Hello  @huand

This issue can be resolved with the latest Humble-develop branch.
Please let me know if there are any further issues.
This issue will be closed.

closed 2025-02-19 by doosan-robotics