DoosanRobotics/doosan-robot2 #45: Moveit failed to build

Dorianproject, 2024-03-21:
I'm trying to install moveit with your instruction and it fails always with the build.
for: rosdep install -r --from-paths src --ignore-src --rosdistro foxy -y
I get: Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

I tried to build and it said first that the eigenstl-container was missing. i installed it with: 
sudo apt install libeigen-stl-containers-dev
then random_numbers was missing i installed with : sudo apt install ros-foxy-random-numbers
Now the new issue is :
  Could not find a package configuration file provided by "fcl" with any of
  the following names:
    fclConfig.cmake
    fcl-config.cmake
________________________________
"sudo apt install libfcl-dev "
said:
libfcl-dev is already the newest version (0.5.0-5build1).
The following package was automatically installed and is no longer required:
  libpkgconf3
________________________________
How can i fix this and is there maybe a full guide how to get moveit running?
Using Ubuntu 20.04 with Ros2 Foxy

SakethTecholution, 2024-05-31:
@Dorianproject Could u be able resolve this error

doosan-robotics, 2025-02-19:
Hello @Dorianproject 

This issue can be resolved with the latest Humble-develop branch.
Please let me know if there are any further issues.
This issue will be closed.

closed 2025-02-19 by doosan-robotics