DoosanRobotics/doosan-robot2 #46: Root link

21spike21, 2024-03-21:
"Hi, I'm trying to calculate the bias force of the h2515.blue robot. I've taken the URDF model, but when I try to calculate it, the software asks me to input the root link. However, I don't understand what the root link is. I've tried to write 'base', but it doesn't work. So, I'm not sure if it requires an array of links or just the root. Can someone please help me? Thank you very much

Dorianproject, 2024-03-22:
Maybe try "base_0" or  "world" since the trajectory goes like world --> base_0 ...

21spike21, 2024-03-22:
my file start with
<?xml version="1.0" ?>
<robot name="h2515">

  
  <link name="base"/>
  <joint name="fixed" type="fixed">
    <parent link="base"/>
    <child link="base_0"/>
  </joint>

21spike21, 2024-03-22:
<?xml version="1.0" ?>
<robot name="h2515">

  
  <link name="base"/>
  <joint name="fixed" type="fixed">
    <parent link="base"/>
    <child link="base_0"/>
  </joint>

21spike21, 2024-03-22:
my urdf file start with
<?xml version="1.0" ?>
<robot name="h2515">

  
  <link name="base"/>
  <joint name="fixed" type="fixed">

21spike21, 2024-03-22:
my urdf file start with
<?xml version="1.0"?>
<robot name="h2515">

<link name="base"/>
<joint name="fixed" type="fixed">

closed 2024-03-22 by 21spike21

21spike21, 2024-03-22:
[image]

Dorianproject, 2024-03-22:
Look into the xacro folder. There should be a general file for your robot : " h2515.urdf.xacro" and a specific file called: " macro.h2515.blue.xacro " there you can see where your trajectory really starts and where it ends. If you want to implement 3D files or change something do it there. I tryed with the URDF but it didnt effect my robot. Xacro worked for me

21spike21, 2024-03-22:
the file h2515.urdf.xacro contains 
[image]

21spike21, 2024-03-22:
the file h2515.blue.urdf
[image]

Dorianproject, 2024-03-22:
Yes so its like i said the root should be world or base_0

21spike21, 2024-03-22:
but if I use base_0 it doesn't find
[image]