DoosanRobotics/doosan-robot2 #48: Error running the dsr_example_py files

MLDario, 2024-03-28:
I'm trying to run the python files in the dsr_example, yesterday they worked, today not anymore.
This is the error i got:
rosrun dsr_example_py dance_m1013.py 

_robot_id =dsr01
_robot_model =m1013
_srv_name_prefix =/dsr01m1013
_topic_name_prefix =/dsr01m1013
Traceback (most recent call last):
  File "/home/dario/doosan_final_ws/src/doosan-robot/dsr_example/py/scripts/demo/dance_m1013.py", line 116, in <module>
    movej(JReady, v=20, a=20)
  File "/home/dario/doosan_final_ws/src/doosan-robot/common/imp/DSR_ROBOT.py", line 1456, in movej
    ret = _movej(pos, vel, acc, time, radius, mod, ra, v, a, t, r, _async=0)
  File "/home/dario/doosan_final_ws/src/doosan-robot/common/imp/DSR_ROBOT.py", line 1554, in _movej
    srv = _ros_movej(_pos, _vel[0], _acc[0], _time, _radius, mod, ra, _async)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 502, in call
    service_uri = self._get_service_uri(request)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 474, in _get_service_uri
    rospy.core.parse_rosrpc_uri(self.uri)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/core.py", line 113, in parse_rosrpc_uri
    if uri.startswith(ROSRPC):
AttributeError: 'NoneType' object has no attribute 'startswith'
shutdown time!
shutdown time!
shutdown time!

I tried to uninstall and install everything again several times, but didn't work. 
I also got an error in the masterapi.py file and I had to comment the next row, in the lookupService function:
        #return self._succeed(self.handle.lookupService(self.caller_id, service))
I don't know if this can be the problem.

Thanks very much for the attention, hope someone can help me.

song-ms, 2024-04-03:
Hello @MLDario , 

Thank you for your interest and for using our ROS services. 
This repository contains Q&A related to Ros2. 
While the Ros2 version does not provide dance.py in the examples, could you please let us know which version you are currently using? 

Thank you.

MLDario, 2024-04-03:
Hello @song-ms,

Thank you very much for your reply.
I'm sorry to say that I posted the question in the wrong page because i'm actually using Ros1 and not Ros2.

I really hope that the problem can still be solved.

Thank you.

closed 2025-02-19 by doosan-robotics