DoosanRobotics/doosan-robot2 #53: ROS2 HUMBLE

SakethTecholution, 2024-04-21:
I had 2 doosan robots. I want to use ros2 but all of the packages are outdated since foxy distrubution is at EOL , Humble is released 2 years back . I dont know why doosan packages are these much outdated for 2 years.
@song-ms @doosan-robotics @woawo1213 
Can you please let us know if we can still use foxy or use ros1 itself(Noetic) . what is your suggestion into it

Dorianproject, 2024-04-21:
You can use Foxy. I use Foxy in a docker container and the packages just work fine. The Doosan packages are currently being updated to Humble but they didn't worked for me so far. I get a lot of building errors etc..

SakethTecholution, 2024-04-21:
@Dorianproject Ok Thank you

SakethTecholution, 2024-04-21:
@song-ms Can i know your suggestion too on this and when can we expect the entire packages in ros2 humble

song-ms, 2024-05-10:
Dear @SakethTecholution , 

Thank you for your interest. We are currently in the process of developing and testing the full package with a humble target date of June 1st. Your patience is greatly appreciated.

Thank you

closed 2024-10-08 by doosan-robotics