DoosanRobotics/doosan-robot2 #56: Force applied in virtual mode

pfcouto, 2024-05-27:
Hello, I was wondering if I place an object and made the robot move the robot in any way if it is possible to know the force being applied by the robot in kg or in any other metrics.

Thank you!

leeminju531, 2025-04-25:
Sorry for the late response.

The necessary data might be available [here](https://github.com/DoosanRobotics/doosan-robot2/blob/89bcbc24b6761fee95859a5bc25edac17910a71f/dsr_hardware2/src/dsr_hw_interface2.cpp#L429), but we don’t explicitly publish it to avoid adding overhead to the system. However, you can customize and access it within the controller if needed.

leeminju531, 2025-04-25:
Oh, we don't support rt connection at virtual mode.

closed 2025-04-29 by song-ms