DoosanRobotics/doosan-robot2 #61: Humble and gz_ros2_control building error

Dorianproject, 2024-06-05:
I cant build the gz_ros2_control pkg. I get these errors:

```
--- stderr: gz_ros2_control
CMake Error at CMakeLists.txt:27 (find_package):
  By not providing "Findgz_sim_vendor.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz_sim_vendor", but CMake did not find one.

  Could not find a package configuration file provided by "gz_sim_vendor"
  with any of the following names:

    gz_sim_vendorConfig.cmake
    gz_sim_vendor-config.cmake

  Add the installation prefix of "gz_sim_vendor" to CMAKE_PREFIX_PATH or set
  "gz_sim_vendor_DIR" to a directory containing one of the above files.  If
  "gz_sim_vendor" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< gz_ros2_control [2.36s, exited with code 1]
```
Also on there git it says there is no humble version for gz_sim_vendor etc. so why is it in the guide?

gauthammsam57, 2024-06-20:
Which version of gazebo are you using? 
I had this issue, cause I did not install gazebo. 

I installed gazebo fortress(ROS installation) and then rebooted the machine and tried again. That fixed my problem.

asher-williams-techolution, 2024-06-22:
If you installed gazebo11 through the gazebo11-gz-cli ppa then you will need to remove it with `sudo add-apt-repository --remove ppa:openrobotics/gazebo11-gz-cli`

Update your lists:
`sudo apt update`

And then install this version of the gazebo11 library
`sudo apt install libgazebo11=11.10.2+dfsg-1`

From there `rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y` should install the rest of the dependencies for you.

Atleast all of this worked for me.

doosan-robotics, 2025-02-19:
This issue can be resolved with the latest Humble version.
Please let me know if there are any further issues.
Otherwise, this issue will be closed.

closed 2025-02-19 by doosan-robotics