DoosanRobotics/doosan-robot2 #63: TCP and  TOOL setting issues from  ROS

Arpit-Bhat, 2024-06-10:
1. I set and enable the Tool setting in Teach Pendent, then I try to get current TCP feedback in ROS environment, and it shows the current set TCP, but when I disconnect and reconnect the ROS master and control launch file, it automatically disable the TOOL setting in Teach Pendent.
2. Now, I try to use following command to create TCP or TOOL and  try to set the TCP and TOOL but its shown the error index 7170
     1. ConfigCreateTcp.srv
     2. SetCurrentTcp.srv
     3. ConfigCreateTool.srv
     4. SetCurrentTool.srv

Command in python file
from dsr_msgs.srv import ConfigCreateTcp
from dsr_msgs.srv import SetCurrentTcp

    create_tcp = rospy.ServiceProxy("/dsr01a0509/tcp/config_create_tcp",ConfigCreateTcp)
    create_tcp(name="tt2",pos=[0.0, 0.0, 170, 0.0, 0.0, 0.0])
    print(create_tcp)

    ros_set_current_tcp = rospy.ServiceProxy("/dsr01a0509/tcp/set_current_tcp",SetCurrentTcp)
    ros_set_current_tcp("tt2")
    print(ros_set_current_tcp)

Output of python file
_robot_id =dsr01
_robot_model =a0509
_srv_name_prefix =/dsr01a0509
_topic_name_prefix =/dsr01a0509
<rospy.impl.tcpros_service.ServiceProxy object at 0x7f7d6f9b3d30>
<rospy.impl.tcpros_service.ServiceProxy object at 0x7f7d6f9b3eb0>
([0.2060876190662384, -1.2643894251596066e-06, 1055.5, 179.99964904785156, -0.012926306575536728, -179.99986267089844], 0)
shutdown time!
shutdown time!
shutdown time!

Master node responce
[ INFO] [1718016769.336561224]: < get_current_posx_cb >
[ WARN] [1718017058.862818040]: [callback OnLogAlarm]
[ WARN] [1718017058.862898980]:  level : 2
[ WARN] [1718017058.862958767]:  group : 5
[ WARN] [1718017058.863041427]:  index : 7170
[ WARN] [1718017058.863106249]:  param : 1
[ WARN] [1718017058.863183053]:  param : 
[ WARN] [1718017058.863538991]:  param : 
[ WARN] [1718017058.866212921]: [callback OnLogAlarm]
[ WARN] [1718017058.866293947]:  level : 2
[ WARN] [1718017058.866373624]:  group : 5
[ WARN] [1718017058.866439583]:  index : 7170
[ WARN] [1718017058.866502957]:  param : 0
[ WARN] [1718017058.866589667]:  param : 
[ WARN] [1718017058.866682349]:  param :

Arpit-Bhat, 2024-06-11:
I briefly describe the issue in ROS 1 package

leeminju531, 2024-12-12:
Dear, @Arpit-Bhat 

> 1. I set and enable the Tool setting in Teach Pendent, then I try to get current TCP feedback in ROS environment, and it shows the current set TCP, but when I disconnect and reconnect the ROS master and control launch file, it automatically disable the TOOL setting in Teach Pendent. Moreover, when I kill the control launch file , the Tool setting in teach pendant also disabled

Similarly, I have noticed that tool service is stored as volatile when accessed in ROS2. 
I believe that Tool and TCP settings need to be handled at the hardware interface and attached to the controller frameworks to reduce confusion. I will open follow-up issues to address this case.

> 2. Now, I try to use the following command to create a TCP or TOOL, and I also try to set the TCP and TOOL, but it shows an error index of 7170, hence the following commands are not working.
>    
>    1. ConfigCreateTcp.srv
>    2. SetCurrentTcp.srv
>    3. ConfigCreateTool.srv
>    4. SetCurrentTool.srv
> 
@Bapsti 
Please set the robot mode into manual mode (0), which is generally used for robot configuration and hand guide operation on the teach pendent. you can use the following command :  
`ros2 service call /dsr01/system/set_robot_mode dsr_msgs2/srv/SetRobotMode "robot_mode: 0"` 

F/w : https://github.com/doosan-robotics/doosan-robot/issues/208

cc: @song-ms

doosan-robotics, 2025-02-19:
Please let me know if there are any further issues.
Otherwise, this issue will be closed.

closed 2025-02-19 by doosan-robotics