DoosanRobotics/doosan-robot2 #65: *humble-devel* Moveit plan gets successful for execute but does not moving the cobot.

gauthammsam57, 2024-06-21:
I have successful in installing the pkg and start the rviz.

I made modifications to work for my use case by changing the hard coded `m1013` to `a0509` in moveit launch file [MoveItConfigsBuilder](https://github.com/doosan-robotics/doosan-robot2/blob/81e8b7e75f6cfaec680da85e1620d3446af91f42/dsr_bringup2/launch/dsr_bringup2_moveit.launch.py#L111).

To start the move_group. I also added `run_move_group_node` to the LaunchDescription. 

Here are the changes I did
[image]

Currently in `RViz2` I am able to plan the action to move the doosan arm. But when I click execute it shows successful but the robot arm does not move. 

`ros2 launch dsr_bringup2 dsr_bringup2_moveit.launch.py mode:=real host:=192.168.1.124 model:=a0509 color:=blue`

```txt
[INFO] [launch]: All log files can be found below /home/cobot/.ros/log/2024-06-21-19-52-42-609266-cobot-357766
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [connection-1]: process started with pid [357769]
[INFO] [ros2_control_node-2]: process started with pid [357771]
[INFO] [move_group-3]: process started with pid [357773]
[INFO] [robot_state_publisher-4]: process started with pid [357775]
[INFO] [spawner-5]: process started with pid [357777]
[INFO] [spawner-6]: process started with pid [357787]
[ros2_control_node-2] [WARN] [1718979763.906330572] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-2] [INFO] [1718979763.906998191] [resource_manager]: Loading hardware 'a0509' 
[ros2_control_node-2] [INFO] [1718979763.916534592] [resource_manager]: Initialize hardware 'a0509' 
[robot_state_publisher-4] [INFO] [1718979764.028389807] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1718979764.028570093] [robot_state_publisher]: got segment link_1
[robot_state_publisher-4] [INFO] [1718979764.028590027] [robot_state_publisher]: got segment link_2
[robot_state_publisher-4] [INFO] [1718979764.028601714] [robot_state_publisher]: got segment link_3
[robot_state_publisher-4] [INFO] [1718979764.028612270] [robot_state_publisher]: got segment link_4
[robot_state_publisher-4] [INFO] [1718979764.028622652] [robot_state_publisher]: got segment link_5
[robot_state_publisher-4] [INFO] [1718979764.028633087] [robot_state_publisher]: got segment link_6
[robot_state_publisher-4] [INFO] [1718979764.028643612] [robot_state_publisher]: got segment world
[move_group-3] Error:   Semantic description is not specified for the same robot as the URDF
[move_group-3]          at line 664 in ./src/model.cpp
[move_group-3] [INFO] [1718979764.058113995] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0897243 seconds
[move_group-3] [INFO] [1718979764.058537725] [moveit_robot_model.robot_model]: Loading robot model 'a0509'...
[connection-1] [INFO] [1718979764.072726084] [connection_node]: mobile: none
[move_group-3] [WARN] [1718979764.224755298] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[move_group-3] [INFO] [1718979764.646548781] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-3] [INFO] [1718979764.646748346] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-3] [INFO] [1718979764.647628146] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-3] [INFO] [1718979764.648133153] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-3] [INFO] [1718979764.648153787] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-3] [INFO] [1718979764.648502147] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-3] [INFO] [1718979764.648519404] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-3] [INFO] [1718979764.648921543] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-3] [INFO] [1718979764.649321938] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-3] [WARN] [1718979764.649665142] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-3] [ERROR] [1718979764.649684961] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[move_group-3] [INFO] [1718979764.875422178] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[move_group-3] [INFO] [1718979764.892367809] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-3] [INFO] [1718979764.895373213] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[move_group-3] [INFO] [1718979764.895390114] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[move_group-3] [INFO] [1718979764.895397137] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[move_group-3] [INFO] [1718979764.895423468] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-3] [INFO] [1718979764.895442430] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[move_group-3] [INFO] [1718979764.895450907] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-3] [INFO] [1718979764.895466541] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-3] [INFO] [1718979764.895473765] [moveit_ros.fix_start_state_collision]: Param 'ompl.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-3] [INFO] [1718979764.895480646] [moveit_ros.fix_start_state_collision]: Param 'ompl.max_sampling_attempts' was not set. Using default value: 100
[move_group-3] [INFO] [1718979764.895494755] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-3] [INFO] [1718979764.895501111] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[move_group-3] [INFO] [1718979764.895506087] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-3] [INFO] [1718979764.895510989] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-3] [INFO] [1718979764.895515640] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[move_group-3] [INFO] [1718979764.895552481] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[move_group-3] [INFO] [1718979764.898609561] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'chomp'
[move_group-3] [ERROR] [1718979764.900612958] [moveit.ros_planning.planning_pipeline]: Exception while loading planner 'chomp_interface/CHOMPPlanner': According to the loaded plugin descriptions the class chomp_interface/CHOMPPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner pilz_industrial_motion_planner/CommandPlannerAvailable plugins: ompl_interface/OMPLPlanner, pilz_industrial_motion_planner/CommandPlanner
[... 213 lines cut ...]
```txt
[move_group-3] [INFO] [1718980080.464306192] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[... cut ...]
[move_group-3] [INFO] [1718980096.944730441] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-3] [INFO] [1718980096.944853324] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: SUCCEEDED
[rviz2-7] [INFO] [1718980096.945617629] [move_group_interface]: Execute request success!
```

Kindly let me know what I am missing to move the robot. 

PS> I am able to move the robot using the ros2 services directly, which means there is no issue in hardware connection or tcp, its just not executing the moveit plan.

codeundercoverdev, 2024-07-08:
Were you able to figure this out? I'm having the same issue.

SakethTecholution, 2024-07-09:
@gauthammsam57 Could be able to figure this issue out.? Is your controller version updated and can you please let us know which controller version ur using and if its figured out please let us know...

gauthammsam57, 2024-07-09:
@SakethTecholution Its still not resolved for me. Even with the latest pull. 
I have updated all the pkgs to the latest. So even the controller version should be latest. 
Could you guide me how to check the Controller version please?

SakethTecholution, 2024-07-09:
@song-ms Can you please look into this issue as we have heavy dependency using Moveit. Please let us know if the problem persists or needs time for resolving , so that we could use DRL for motion planning

SakethTecholution, 2024-07-09:
@gauthammsam57 You can go to settings in teachpendant and go to update . there it will show the controller version 
[image]
In the website doosan mentioned that recent release is only 2..11 but how come in document they mentioned that minimum is required
[image]
@song-ms @doosan-robotics Can you please guide us through this issue as lot of ppl are facing the issue with the current moveit package with A0509. In my case too, i couldnt be able to move the cobot even after the latest pull

SakethTecholution, 2024-07-09:
@gauthammsam57 Please let me if update of controller works.. !

gauthammsam57, 2024-07-09:
@SakethTecholution 
Current Integration Package Version is `GV02110200` 

But I don't the teach pendent version matters. For me to control and use, I generally don't even 'switch on' the teach pendent to control using ROS 2.

I think it could be come other controller. Maybe something from `ros-controls`, not sure?

SakethTecholution, 2024-07-09:
@gauthammsam57 Can you please let me know which ros service call you used to move the robot and what launch file u used. Is it gazebo or movit

gauthammsam57, 2024-07-10:
For me it works on all. basic rviz launch, gazebo and moveit. You can check the service by doing `ros2 service list`
There is `move_joint` and `move_jointx` I use both of them to control the bot. 

Example:
Move all joints to [0,0,0,0,0,0] degree.
```shell
ros2 service call /dsr01/motion/move_joint dsr_msgs2/srv/MoveJoint '{vel: 255.0, acc: 255.0}'
```

PS: Make sure the cobot indication light is white. 
Color | Mode
---|---
White | Auto Control
Blue | Manual Control
Yellow | Recovery mode
Red | Error/Loading

xwn-github, 2024-07-12:
I encountered the same problem, moveit plan succeed, moveit execution also showed success, but the real robot (h2017) did not move
When I started the test, the version of the manipulator controller was version 2.10.3. Later, it was updated to version 2.11.2 and tested again, with the same result.
The doosan ros2 package I pulled is the latest humble branch.

gauthammsam57, 2024-07-12:
@xwn-github: When you said Manipulator Controller Version: Are you talking about the console? Or some pkg version? 

If its some pkg version could you please guide on how to check it?

xwn-github, 2024-07-15:
The Manipulator controller system version, or DART Platform software version

gauthammsam57, 2024-07-30:
@SakethTecholution So far it does not work. There are no commits that fixes this issue. At least as far as I know.

mbonu-luxmp, 2024-10-15:
I have the same issue does any body solved it?

doosan-robotics, 2025-02-19:
Hello @codeundercoverdev and All, 

This issue can be resolved with the latest Humble package
Please let me know if there are any further issues.
This issue will be closed.

closed 2025-02-19 by doosan-robotics