DoosanRobotics/doosan-robot2 #69: Failed to initialize hardware

codeundercoverdev, 2024-07-02:
I am unable to get RViz to work with the humble distributioin on UBuntu 22.04.

I just installed and ran `ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=virtual host:=127.0.0.1 port:=12345 model:=m1013` exactly as stated in the readme. Regardless of the model I decide to use, I always get this message (but with a different model depending on what I use). I've tried removing all the arguments as well, but no matter what I do I get this message:

```
[INFO] [launch]: All log files can be found below /home/phil/.ros/log/2024-07-02-15-36-51-775414-phil-vm-9836
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [connection-1]: process started with pid [9839]
[INFO] [ros2_control_node-2]: process started with pid [9841]
[INFO] [robot_state_publisher-3]: process started with pid [9843]
[INFO] [spawner-4]: process started with pid [9845]
[INFO] [spawner-5]: process started with pid [9847]
[connection-1] [INFO] [1719959812.792919326] [dsr01.connection_node]: @@@@@@@@@@@@@@@@@@@@@@@@@@@ run_drcf
[connection-1] sh: 1: /home/phil/ros2_ws/install/common2/share/common2/bin/run_drcf.sh: Permission denied
[robot_state_publisher-3] [INFO] [1719959812.957447075] [dsr01.robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1719959812.958376456] [dsr01.robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1719959812.958950496] [dsr01.robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1719959812.959295367] [dsr01.robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1719959812.959552543] [dsr01.robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1719959812.959733646] [dsr01.robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1719959812.960064851] [dsr01.robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1719959812.960309546] [dsr01.robot_state_publisher]: got segment world
[ros2_control_node-2] [WARN] [1719959813.073262810] [dsr01.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-2] [INFO] [1719959813.074083484] [resource_manager]: Loading hardware 'm1013' 
[ros2_control_node-2] [INFO] [1719959813.173292150] [resource_manager]: Initialize hardware 'm1013' 
[spawner-4] [INFO] [1719959815.511390054] [dsr01.spawner_dsr_controller2]: Waiting for '/dsr01/controller_manager' services to be available
[spawner-5] [INFO] [1719959815.538801545] [dsr01.spawner_joint_state_broadcaster]: Waiting for '/dsr01/controller_manager' services to be available
[spawner-4] [INFO] [1719959817.582501335] [dsr01.spawner_dsr_controller2]: Waiting for '/dsr01/controller_manager' services to be available
[spawner-5] [INFO] [1719959817.591444426] [dsr01.spawner_joint_state_broadcaster]: Waiting for '/dsr01/controller_manager' services to be available
[spawner-4] [INFO] [1719959819.630133907] [dsr01.spawner_dsr_controller2]: Waiting for '/dsr01/controller_manager' services to be available
[spawner-5] [INFO] [1719959819.639873122] [dsr01.spawner_joint_state_broadcaster]: Waiting for '/dsr01/controller_manager' services to be available
[ros2_control_node-2] [INFO] [1719959821.201391745] [dsr01.dsr_hw_interface2]: name: dsr01
[ros2_control_node-2] [INFO] [1719959821.202044405] [dsr01.dsr_hw_interface2]: rate: 100
[ros2_control_node-2] [INFO] [1719959821.202416888] [dsr01.dsr_hw_interface2]: standby: 5000
[ros2_control_node-2] [INFO] [1719959821.202668636] [dsr01.dsr_hw_interface2]: command: true
[ros2_control_node-2] [INFO] [1719959821.202905667] [dsr01.dsr_hw_interface2]: host: 127.0.0.1
[ros2_control_node-2] [INFO] [1719959821.203123202] [dsr01.dsr_hw_interface2]: port: 12345
[ros2_control_node-2] [INFO] [1719959821.203357101] [dsr01.dsr_hw_interface2]: mode: virtual
[ros2_control_node-2] [INFO] [1719959821.203623837] [dsr01.dsr_hw_interface2]: model: m1013
[ros2_control_node-2] [INFO] [1719959821.203842957] [dsr01.dsr_hw_interface2]: gripper: none
[ros2_control_node-2] [INFO] [1719959821.204061667] [dsr01.dsr_hw_interface2]: mobile: none
[ros2_control_node-2] [INFO] [1719959821.204513202] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] [INFO] [1719959821.204631100] [dsr_hw_interface2]:     INITAILIZE
[ros2_control_node-2] [INFO] [1719959821.204634565] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] Connection refused
[ros2_control_node-2] [ERROR] [1719959821.227011599] [dsr_hw_interface2]:     DSRInterface::init() DRCF connecting ERROR!!!
[ros2_control_node-2] [INFO] [1719959821.227053517] [resource_manager]: Failed to initialize hardware 'm1013'
[ros2_control_node-2] [WARN] [1719959821.227058762] [resource_manager]: System hardware component 'm1013' from plugin 'dsr_hardware2/DRHWInterface' failed to initialize.
[ros2_control_node-2] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-2]   what():  Wrong state or command interface configuration.
[ros2_control_node-2] missing state interfaces:
[ros2_control_node-2] ' joint_1/position '	' joint_1/velocity '	' joint_2/position '	' joint_2/velocity '	' joint_3/position '	' joint_3/velocity '	' joint_4/position '	' joint_4/velocity '	' joint_5/position '	' joint_5/velocity '	' joint_6/position '	' joint_6/velocity '	
[ros2_control_node-2] missing command interfaces:
[ros2_control_node-2] ' joint_1/position '	' joint_1/velocity '	' joint_2/position '	' joint_2/velocity '	' joint_3/position '	' joint_3/velocity '	' joint_4/position '	' joint_4/velocity '	' joint_5/position '	' joint_5/velocity '	' joint_6/position '	' joint_6/velocity '	
[ros2_control_node-2] Stack trace (most recent call last):
[ros2_control_node-2] #15   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-2] #14   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x5ec117d0ed84, in 
[ros2_control_node-2] #13   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x712303029e3f]
[ros2_control_node-2] #12   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x712303029d8f]
[ros2_control_node-2] #11   Object "/opt/ros/humble/lib/controller_manager/ros2_control_node", at 0x5ec117d0e89e, in 
[ros2_control_node-2] #10   Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x712303a0ee79, in controller_manager::ControllerManager::ControllerManager(std::shared_ptr<rclcpp::Executor>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&)
[ros2_control_node-2] #9    Object "/opt/ros/humble/lib/libcontroller_manager.so", at 0x712303a0a6bf, in controller_manager::ControllerManager::init_resource_manager(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ros2_control_node-2] #8    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x712303672f1c, in hardware_interface::ResourceManager::load_urdf(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool)
[ros2_control_node-2] #7    Object "/opt/ros/humble/lib/libhardware_interface.so", at 0x71230364f5dd, in 
[ros2_control_node-2] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7123034ae4d7, in __cxa_throw
[ros2_control_node-2] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7123034ae276, in std::terminate()
[ros2_control_node-2] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7123034ae20b, in 
[ros2_control_node-2] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7123034a2b9d, in 
[... 37 lines cut ...]
[connection-1] sh: 1: docker: not found
[connection-1] Exception ignored in: <function VirtualDRCF.__del__ at 0x7a69c9075900>
[connection-1] Traceback (most recent call last):
[connection-1]   File "/home/phil/ros2_ws/install/dsr_bringup2/lib/python3.10/site-packages/dsr_bringup2/connection.py", line 19, in __del__
[connection-1]     self.terminate_drcf()
[connection-1]   File "/home/phil/ros2_ws/install/dsr_bringup2/lib/python3.10/site-packages/dsr_bringup2/connection.py", line 33, in terminate_drcf
[connection-1]     rclpy.shutdown()
[... cut ...] 5cba --params-file /tmp/launch_params_uxx0d6xl --params-file /tmp/launch_params_59g80hgv --params-file /tmp/launch_params_rr061uvk --params-file /tmp/launch_params_ojzgmou1 --params-file /tmp/launch_params_w_t1sflu --params-file /tmp/launch_params_axrq3gqv'].
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 9843]
[INFO] [rviz2-6]: process has finished cleanly [pid 9904]
```

The Robot Model in the RVIZ GUI seems to reiterate what the console says:

[image]

How can I fix this?

gauthammsam57, 2024-07-05:
If you are trying to run virtual. I would recommend using the gazebo launch file. (`ros2 launch dsr_bringup2 dsr_bringup2_gazebo.launch.py mode:=virtual model:=m1013 host:=127.0.0.1`)

Also you need to have docker and install emulator in your machine. 

If you want rviz then try to run with param `gui:=true`

marknabil, 2024-10-04:
I have the same issue, I already run `install_emulator.sh`  before the ros2 launch

doosan-robotics, 2025-02-19:
Hello @marknabil 

This issue can be resolved with the latest Humble package
Please let me know if there are any further issues.
This issue will be closed.

closed 2025-02-19 by doosan-robotics