DoosanRobotics/doosan-robot2 #81: Can't find "move" operation in service list

hyeonjjjin, 2024-07-24:
Hello,

I am following a tutorial video for ROS2, where after running ros2 service list, I expect to see commands related to "move" as shown in the video. However, when I run the command, I only see the following list:

----------------------------------------------------------------------------------------------
$ ros2 service list | grep dsr01

/dsr01/connection_node/describe_parameters
/dsr01/connection_node/get_parameter_types
/dsr01/connection_node/get_parameters
/dsr01/connection_node/list_parameters
/dsr01/connection_node/set_parameters
/dsr01/connection_node/set_parameters_atomically
/dsr01/gazebo_connection/describe_parameters
/dsr01/gazebo_connection/get_parameter_types
/dsr01/gazebo_connection/get_parameters
/dsr01/gazebo_connection/list_parameters
/dsr01/gazebo_connection/set_parameters
/dsr01/gazebo_connection/set_parameters_atomically
/dsr01/gz/controller_manager/configure_controller
/dsr01/gz/controller_manager/describe_parameters
/dsr01/gz/controller_manager/get_parameter_types
/dsr01/gz/controller_manager/get_parameters
/dsr01/gz/controller_manager/list_controller_types
/dsr01/gz/controller_manager/list_controllers
/dsr01/gz/controller_manager/list_hardware_components
/dsr01/gz/controller_manager/list_hardware_interfaces
/dsr01/gz/controller_manager/list_parameters
/dsr01/gz/controller_manager/load_controller
/dsr01/gz/controller_manager/reload_controller_libraries
/dsr01/gz/controller_manager/set_hardware_component_state
/dsr01/gz/controller_manager/set_parameters
/dsr01/gz/controller_manager/set_parameters_atomically
/dsr01/gz/controller_manager/switch_controller
/dsr01/gz/controller_manager/unload_controller
/dsr01/gz/dsr_position_controller/describe_parameters
/dsr01/gz/dsr_position_controller/get_parameter_types
/dsr01/gz/dsr_position_controller/get_parameters
/dsr01/gz/dsr_position_controller/list_parameters
/dsr01/gz/dsr_position_controller/set_parameters
/dsr01/gz/dsr_position_controller/set_parameters_atomically
/dsr01/gz/gz_ros2_control/describe_parameters
/dsr01/gz/gz_ros2_control/get_parameter_types
/dsr01/gz/gz_ros2_control/get_parameters
/dsr01/gz/gz_ros2_control/list_parameters
/dsr01/gz/gz_ros2_control/set_parameters
/dsr01/gz/gz_ros2_control/set_parameters_atomically
/dsr01/gz/joint_state_broadcaster/describe_parameters
/dsr01/gz/joint_state_broadcaster/get_parameter_types
/dsr01/gz/joint_state_broadcaster/get_parameters
/dsr01/gz/joint_state_broadcaster/list_parameters
/dsr01/gz/joint_state_broadcaster/set_parameters
/dsr01/gz/joint_state_broadcaster/set_parameters_atomically
/dsr01/gz/robot_state_publisher/describe_parameters
/dsr01/gz/robot_state_publisher/get_parameter_types
/dsr01/gz/robot_state_publisher/get_parameters
/dsr01/gz/robot_state_publisher/list_parameters
/dsr01/gz/robot_state_publisher/set_parameters
/dsr01/gz/robot_state_publisher/set_parameters_atomically
/dsr01/robot_state_publisher/describe_parameters
/dsr01/robot_state_publisher/get_parameter_types
/dsr01/robot_state_publisher/get_parameters
/dsr01/robot_state_publisher/list_parameters
/dsr01/robot_state_publisher/set_parameters
/dsr01/robot_state_publisher/set_parameters_atomically
----------------------------------------------------------------------------------------------

It seems like the "move" commands are missing. Could someone please assist me with this issue? 
Any guidance would be greatly appreciated.

Thank you.

SakethTecholution, 2024-07-29:
@hyeonjjjin Run gazebo with gui:=True or else run rviz paralley , then you can see that service call are availble. I am also not sure why it is like this , but this got helped

hyeonjjjin, 2024-07-30:
Thank you for your response.

I have resolved the issue; the reason motion could not be found in my environment was due to an installation problem.
The command
`$ rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`
was not executed properly, which caused the motion feature to not appear.

After reinstalling with the command
`$ sudo rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`

I was able to find the motion command.

doosan-robotics, 2025-02-19:
Hello @hyeonjjjin 

Please let me know if there are any further issues.
This issue will be closed.

closed 2025-02-19 by doosan-robotics