DoosanRobotics/doosan-robot2 #83: Nvidia Issac sim

SakethTecholution, 2024-07-30:
@song-ms @doosan-robotics @bmdyrdal @leeminju531 
Can you please let us know what could be the update on Nvidea Issac sim support since in our project we are using lot of RL projects with doosan a0509 which is dependent on the use of Issac sim and Issac labs. Can you please let when can we get the Issac Ros support from doosan or atleast let us know the ongoing process so that we can be aware of deadlines and set our project accordingly

leeminju531, 2024-07-31:
@SakethTecholution
AFAIK, other teams are likely to engage in use of issac sim. according to them, we might be assinged to take charge of it (though i'm not sure).
Please let me know if I missed any details.  @song-ms

doosan-robotics, 2025-02-19:
@SakethTecholution 

Thank you for your interest.

The timeline will be shared after an internal review. This year, we are preparing to add the new P3020 model, release the Jazzy version, and provide various ROS package updates. We appreciate your continued interest and support.

closed 2025-02-19 by doosan-robotics