DoosanRobotics/doosan-robot2 #86: Manual Jog service is not working with set_robot_mode service in ROS2

SakethTecholution, 2024-09-02:
**I want to manually jog the robot while i have doosan ros2 control . But the proviced service is not working in ROS2**

```
doosan-robotics@saketh-Legion-Slim-7-16IRH8:/workspace/doosan_ws$ ros2 service call /dsr01/dsr_test/system/set_robot_mode dsr_msgs2/srv/SetRobotMode "robot_mode: 0" 
waiting for service to become available...
requester: making request: dsr_msgs2.srv.SetRobotMode_Request(robot_mode=0)

response:
dsr_msgs2.srv.SetRobotMode_Response(success=False)

doosan-robotics@saketh-Legion-Slim-7-16IRH8:/workspace/doosan_ws$ ros2 service call /dsr01/system/get_robot_mode dsr_msgs2/srv/GetRobotMode 
waiting for service to become available...
requester: making request: dsr_msgs2.srv.GetRobotMode_Request()

response:
dsr_msgs2.srv.GetRobotMode_Response(robot_mode=1, success=True)

doosan-robotics@saketh-Legion-Slim-7-16IRH8:/workspace/doosan_ws$ ros2 service call /dsr01/dsr_test/system/set_robot_mode dsr_msgs2/srv/SetRobotMode "robot_mode: 2" 
requester: making request: dsr_msgs2.srv.SetRobotMode_Request(robot_mode=2)

response:
dsr_msgs2.srv.SetRobotMode_Response(success=False)
```

SakethTecholution, 2024-09-24:
@song-ms @doosan-robotics can you please help me out into this ..? I am trying to contact doosan team from long team but no support is given till now . Atleast if you can guide me what are the possible issues , so that we can debug and also help the repository to find bugs. But I am not sure why doosan team doesnt responds to the issues

leeminju531, 2024-09-25:
@SakethTecholution 
Sorry for the late response.

I wonder if "manually jog" means handling the robot directly by pushing buttons on the robot's endpoint? If not, could you specify it in more detail?

Jog services typically work in manual mode.

```
doosan-robotics@saketh-Legion-Slim-7-16IRH8:/workspace/doosan_ws$ ros2 service call /dsr01/system/get_robot_mode dsr_msgs2/srv/GetRobotMode 
waiting for service to become available...
requester: making request: dsr_msgs2.srv.GetRobotMode_Request()
```
But currently, your system is in autonomous mode. It needs to be switched to manual mode (maybe the light will turn blue).

refer to : 
```c++
typedef enum {
    ROBOT_MODE_MANUAL,
    ROBOT_MODE_AUTONOMOUS,
    ROBOT_MODE_RECOVERY,
    ROBOT_MODE_BACKDRIVE,
    ROBOT_MODE_MEASURE,
    ROBOT_MODE_INITIALIZE,
    ROBOT_MODE_LAST,
} ROBOT_MODE;
```

The service seems to be named `/dsr01/dsr_test/system/set_robot_mode`, 
```
kspace/doosan_ws$ ros2 service call /dsr01/dsr_test/system/set_robot_mode dsr_msgs2/srv/SetRobotMode "robot_mode: 0" 
waiting for service to become available...
requester: making request: dsr_msgs2.srv.SetRobotMode_Request(robot_mode=0)

response:
dsr_msgs2.srv.SetRobotMode_Response**(success=False)**
```
but when I try to launch with: `ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=real host:=192.168.137.100 port:=12345 model:=m1013`  the service server is available at `/dsr01/system/set_robot_mode`.

Could you try it on the current commits?

cc: @song-ms @doosan-robotics

closed 2025-02-19 by doosan-robotics