DoosanRobotics/doosan-robot2 #87: Cobot Timeout Issue – Random `timeout` in Active and Idle States

gautham-nf, 2024-09-09:
## Description
I am encountering an issue with the cobot timing out unpredictably when using the ROS2 package. The issue occurs under the following conditions:
- While the cobot is actively moving or executing commands, it times out abruptly.
- Even when the cobot is in an idle state (not receiving commands), the connection randomly times out.

There does not appear to be any identifiable pattern or cause for the timeouts. This behavior happens whether I am actively using the cobot or if it is idle.

## System Information
Cobot Model: Doosan A0509
Ethernet Connection: Both PC and Cobot Controller are connected via Ethernet (wired).

## Command Used:
The command I use to launch the connection is as follows:
```shell
ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=real host:=192.168.0.10 model:=a0509 color:=blue
```
## Troubleshooting Attempts:
I have tried the following troubleshooting steps without success:
1. Restarting the ROS2 package to re-establish the connection.
2. Checking the Ethernet connection and ensuring stable network conditions.
3. Reviewing timeout parameters in the package, but no obvious configuration issues were found.

Despite these efforts, the issue persists, and the connection randomly times out during operation or idle periods.

## Error logs
[image]

Any insights or suggestions on how to debug the issue would be greatly appreciated. This is affecting both the active use of the cobot and its idle state, and I haven’t been able to identify any clear cause for the timeouts.

leeminju531, 2024-09-27:
@gautham-nf, Could you provide your host os and branch name?

gautham-nf, 2024-09-27:
OS: Ubuntu
Branch: Humble-devel

I was able to pin point how this error occurs. 

If I connect my computer and the controller over a network switch, then the problem occurs once every 5-10 mins randomly. But if I am connected using a router then this error does not happen. 

Not sure how there is any communication difference between these two type of connection.

doosan-robotics, 2025-02-19:
Hello @gautham-nf 

This issue can be resolved with the latest Humble package.
Please proceed with updating humble source to the new.

doosan-robotics, 2025-02-24:
Hello @gautham-nf 

This issue will be automatically closed as there has been no response from the issue reporter for over 30 days.
If you have any additional inquiries, please create a new issue.
Thank you.

closed 2025-02-24 by doosan-robotics