DoosanRobotics/doosan-robot2 #88: Problems Connecting to Real Robot On Ros2 Foxy with Doosan M1013

jomon250, 2024-09-10:
I am trying to connect with my real robot M1013 and configured the network like the following:
robot_ip:= 192.168.137.100
laptop_ip:= 192.168.137.99 (subnet_mask 255.255.255.0). since the ping-test worked out, I assume the network is connected.

Nevertheless, I do not get anything published on /joint_states, so I'll get the following error message when I launch the respective file:
ros2 launch dsr_launcher2 single_robot_rviz.launch.py mode:=real host:=192.168.137.100  model:=m1013 port:=12345

rviz2-3] Warning: Invalid frame ID "link6" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.14/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "link1" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/b

Do I have to configure anything in the teach pendant ? Thank you in advance!

leeminju531, 2024-09-20:
Maybe there is no need for configuration in the teach pendant.
Could you test it on the `humble-devel` branch? We're working to support the available features there
If the same problems occur, please show the full logs.
Thanks!

doosan-robotics, 2025-02-19:
If there are any further issues, please let me know.
I will proceed with closing this issue.

closed 2025-02-19 by doosan-robotics