DoosanRobotics/doosan-robot2 #93: Safety issue level 2 group 5 index 7056 using torque_rt

SwonGao, 2024-09-24:
Hi, We are using the torque_rt function to use our impedance-based controller. It always raise warnings level2 group 5 index 7056.
When moving, after a few second and motion, it raise the error and trigger the safety mechanism. 
We try to increase the joint vel and acc limit and also filtered the input force, but none of them work well.
Can you show us how to deal with this kind of error? thanks

matteodv99tn, 2024-09-24:
I am working with @SwonGao 
The main problem appears to be the _joint angle tolerance_. We are able to use the ``torque_rt`` function to send the desired torque, and the robot starts moving accordingly.
However when the motion exceeds the +/- 3° angle on one of the joint, the safety stop is triggered with the error indexed 7056.
We tried to get a look at the APIs, but we have not been able to update the target joint configuration of the robot to avoid the triggering of such error.

elton-choi, 2024-10-01:
5.7056 error means SOS Violation Error which is occurred when the robot moves in standstill mode.
refer to this command.
https://manual.doosanrobotics.com/en/api/1.29/Publish/cdrflex-set_safety_mode

Before you move robot, you have to change robot mode and safety mode.
The example in autonomou mode 

Drfl.set_robot_mode(ROBOT_MODE_AUTONOMOUS);
Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE);

Unless, the safety board triggers the fault error like 5.7056.
You can find similar command in ros2.

matteodv99tn, 2024-10-03:
Thank you for your reply, we had been able to have the RT control working in ROS2.
For the future, is there a reference in which all error codes are reported and explained?

Thanks!

doosan-robotics, 2025-02-24:
@matteodv99tn 

You can check the error codes related to troubleshooting in the Dart Studio program.

For more details, please visit:
[Doosan Robotics Resource Center](https://robotlab.doosanrobotics.com/en/board/Resources/Software)

This issue will be closed. If you have any inquiries, please feel free to ask in our issue channel.

Thank you

closed 2025-02-24 by doosan-robotics

hvtung26, 2025-02-25:
Hello, I have the same issue on Windows, after changing from ROS2 to Windows, using movel command,  It always raise warnings level2 group 5 index 7056. Although is has no problems when moving the same position using pendant. Could you please show me what should I do for my issue?