DoosanRobotics/doosan-robot2 #97: Services not available ?

Dorianproject, 2024-10-02:
Does anyone have the problem that no service is available? For me my scripts worked for foxy and early humble packages, now i nothing is working, in my skript dsr_msgs2.srv is found but still no services available when i try to go for the Jog or MoveHome srv its not working.

marknabil, 2024-10-05:
I have an issue with compliance service not available !

Dorianproject, 2025-02-24:
Well turns out you have to change the path from 'system/get_current_pose'
 to '/dsr01/system/get_current_pose' . In foxy there was no problem with that but i understand why it has to be done like this, it would be very helpfull when changes like this would be commented somewhere.

closed 2025-02-24 by Dorianproject