DoosanRobotics/doosan-robot2 #101: readme videos not publically availabe

SakethTecholution, 2024-10-04:
@uzchan98 @doosan-robotics 
https://github.com/doosan-robotics/doosan-robot2/blob/humble-devel/dsr_example2/visualservoing/README.md?plain=1
in the readme of visual servoing which is recently commited, the videos or draw.io is not publically avialble . can you please correct it . I am just curious on how visual looks like in video

SakethTecholution, 2024-10-07:
cc @song-ms

song-ms, 2024-10-08:
@SakethTecholution 

Thank you for letting mo know. we updated readme.

SakethTecholution, 2024-10-08:
@song-ms
Thankyou for the reply 
I have 2 questions with dsr_example 
1. Can visual servoing code will work with real robot and how can I perform callibration with the robot. Can you suggest if there any packages available 
2. Is the real time control package also works with A0509 (since it's mentioned it is used m1013) 

It would be helpful if you can answer these 🙏

SakethTecholution, 2024-10-08:
Cc @doosan-robotics @leeminju531 @bmdyrdal @uzchan98 @Chemx3937 @Mincheol710313

song-ms, 2024-10-11:
@SakethTecholution 

1. This is an example demonstrating visual servoing by following coordinates based on markers. Since calibration is not included in this example, we recommend using the hand-eye calibration package from ROS community.

2. The real-time example is independent of the model.

P.S. Maintenance of the Doosan package is currently being handled by @doosan-robotics, @leeminju531, and @song-ms.

SakethTecholution, 2024-11-01:
These videos are stil not avialble @leeminju531 @doosan-robotics

closed 2025-01-03 by doosan-robotics