DoosanRobotics/doosan-robot2 #104: Multi Jog is not working for orientation using multi jog service call

SakethTecholution, 2024-10-14:
I am trying to use multi jog ros2 service call . it is only working for translation (x,y,z) but for orientation it isnt working

SakethTecholution, 2024-10-14:
@leeminju531 @doosan-robotics @song-ms

SakethTecholution, 2024-10-18:
Could you please respond to the issues? There has been no support from the Doosan team, and much of our development time is being spent identifying whether the problems are on our end or yours. If any further details are needed, I'll provide them promptly, but the delays are impacting progress. Your support would be appreciated.

@doosan-robotics @song-ms 
@leeminju531 @bmdyrdal @uzchan98 @Chemx3937

elton-choi, 2024-10-31:
@SakethTecholution If you want to implement multi jog feature, I recommend you to use speedl function.