DoosanRobotics/doosan-robot2 #107: Intermittent std::runtime_error: "Node '/dsr_hw_interface_update' has already been added to an executor"

gautham-nf, 2024-10-15:
## Description
While using the real cobot, the following error occurs intermittently without a clear pattern. It seems to happen more frequently when I connect to the ROS2 service to move the cobot. The issue terminates the `ros2_control_node` process, interrupting the operation.

## Error Logs
Below are the two variants of the error logs encountered:

### 1. With Stack Trace:
```sh
[ros2_control_node-2] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-2]   what():  Node '/dsr_hw_interface_update' has already been added to an executor.
[ros2_control_node-2] Stack trace (most recent call last) in thread 186766:
[ros2_control_node-2] #20   Object "", at 0xffffffffffffffff, in 
[ros2_control_node-2] #19   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x767a98b2684f]
[ros2_control_node-2] #18   Source "./nptl/pthread_create.c", line 442, in start_thread [0x767a98a94ac2]
[ros2_control_node-2] #17   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x767a98edc252, in 
[ros2_control_node-2] #16   Object "/home/cobot/ros2_ws/install/dsr_hardware2/lib/libdsr_hardware2.so", at 0x767a90ba0745, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> > > > >::_M_run()
[ros2_control_node-2] #15   Object "/home/cobot/ros2_ws/install/dsr_hardware2/lib/libdsr_hardware2.so", at 0x767a90ba14e5, in std::thread::_Invoker<std::tuple<void (*)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> > > >::operator()()
[ros2_control_node-2] #14   Object "/home/cobot/ros2_ws/install/dsr_hardware2/lib/libdsr_hardware2.so", at 0x767a90ba1e30, in void std::thread::_Invoker<std::tuple<void (*)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> > > >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>)
[ros2_control_node-2] #13   Object "/home/cobot/ros2_ws/install/dsr_hardware2/lib/libdsr_hardware2.so", at 0x767a90ba2431, in std::__invoke_result<void (*)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> > >::type std::__invoke<void (*)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> > >(void (*&&)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> >&&)
[ros2_control_node-2] #12   Object "/home/cobot/ros2_ws/install/dsr_hardware2/lib/libdsr_hardware2.so", at 0x767a90ba26eb, in void std::__invoke_impl<void, void (*)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> > >(std::__invoke_other, void (*&&)(sensor_msgs::msg::JointState_<std::allocator<void> > const&), sensor_msgs::msg::JointState_<std::allocator<void> >&&)
[ros2_control_node-2] #11   Object "/home/cobot/ros2_ws/install/dsr_hardware2/lib/libdsr_hardware2.so", at 0x767a90b49ede, in threadFunctionUpdate(sensor_msgs::msg::JointState_<std::allocator<void> > const&)
[ros2_control_node-2] #10   Object "/opt/ros/humble/lib/librclcpp.so", at 0x767a991d8f2e, in rclcpp::spin_some(std::shared_ptr<rclcpp::Node>)
[ros2_control_node-2] #9    Object "/opt/ros/humble/lib/librclcpp.so", at 0x767a991d8e94, in rclcpp::spin_some(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)
[ros2_control_node-2] #8    Object "/opt/ros/humble/lib/librclcpp.so", at 0x767a991cfb1c, in rclcpp::Executor::spin_node_some(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)
[ros2_control_node-2] #7    Object "/opt/ros/humble/lib/librclcpp.so", at 0x767a991a325b, in 
[ros2_control_node-2] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x767a98eae4d7, in __cxa_throw
[ros2_control_node-2] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x767a98eae276, in std::terminate()
[ros2_control_node-2] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x767a98eae20b, in 
[ros2_control_node-2] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x767a98ea2b9d, in 
[ros2_control_node-2] #2    Source "./stdlib/abort.c", line 79, in abort [0x767a98a287f2]
[ros2_control_node-2] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x767a98a42475]
[ros2_control_node-2] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[ros2_control_node-2]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[ros2_control_node-2]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x767a98a969fc]
[ros2_control_node-2] Aborted (Signal sent by tkill() 145758 1000)
[ERROR] [ros2_control_node-2]: process has died [pid 145758, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_fyxv4gw3 --params-file /home/cobot/ros2_ws/install/dsr_controller2/share/dsr_controller2/config/dsr_controller2.yaml'].
```

### 2. Without Stack Trace:
```sh
[ros2_control_node-2] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-2]   what():  Node '/dsr_hw_interface_update' has already been added to an executor.
[ERROR] [ros2_control_node-2]: process has died [pid 59835, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_kge0eh65 --params-file /home/cobot/ros2_ws/install/dsr_controller2/share/dsr_controller2/config/dsr_controller2.yaml'].
```

## Steps to Reproduce
The issue occurs randomly, making it difficult to reproduce consistently. However, it appears more frequently when:
1. Connecting to the ROS2 service to move the cobot.
2. Running with ROS2 Humble.
3. Close the control node. And start the node again. 

## Behavior
The `ros2_control_node` process terminates unexpectedly with a `std::runtime_error` stating:
`Node '/dsr_hw_interface_update' has already been added to an executor`.

## Environment
- **OS**: Ubuntu 22.04.5 LTS
- **ROS Version**: ROS2 Humble
- **Hardware**: Real Cobot

gautham-nf, 2024-11-28:
@leeminju531, @song-ms, @doosan-robotics 
Could you kindly check this out and let me know how to resolve this issue.

leeminju531, 2024-11-28:
@gautham-nf , Sorry for late response.

I think that when restarting launches, some processes or dynamic libraries unexpectedly remain active even after receiving a termination signal.

Could this issue occur even after ensuring all nodes are cleanly terminated?

gautham-nf, 2024-11-28:
@leeminju531. Thanks for the response. 

We have been facing this issue quite a lot. We tried to restart the cobot controller and also the ROS computer. 

Kindly help me to resolve this issue, as this is a major blocker for us to do anything. 

If there is issue with clean termination. Then on restart it should be stable, right?

leeminju531, 2024-11-28:
@gautham-nf ,

Checking whether this occurs after a clean termination seems helpful in figuring it out and addressing the issue!

gautham-nf, 2024-11-28:
Sure @leeminju531, ready to fix the issue any way possible. 

Could you help me on how to terminate cleanly? In my understanding its press `ctrl+c` and wait for it to complete the termination process. 

The best way to avoid any dangling process is by restarting the whole computer and trying out. 

I have tried both and results are still the same. Is there anything specific you would recommend me to try out? 

Also, would like to let you know, this error happens even during cobot operation. That is while the cobot moves the error just ramdomly pops up.

leeminju531, 2024-11-29:
@gautham-nf , 

I believe simply checking the current nodes using `ros2 node list` before starting the launch is sufficient. 
If the issue reappears, please let us know and provide the full logs to help diagnose the problem.

cc: @song-ms @doosan-robotics

gautham-nf, 2024-11-29:
@leeminju531. As per what you said. I was monitoring ros2 node list for dsr package, and found that during operation, the `/dsr_hw_interface_update` node multiplies and sometimes it dies off. 

[image]

Here is the screenshot of the last crash. 
[image]

Right before the crash I noticed there were 3 nodes with name `/dsr_hw_interface_update`

This is soon after the crash
[image]

This means the dsr_* most of the nodes died cleanly and no process should dangling right? 

Kindly me me know whats the next step towards resolving this issue.

gautham-nf, 2024-11-29:
@leeminju531 
On further testing today. I found that for some reason multiple nodes gets created when the CPU is throttled.. And the multiple nodes causes the CPU to throttle more and create some race condition causing it to crash. 

[image]

These are all the ways I could find a way to reproduce the issue. 

All of this is tested by making sure there are no dangling process running and `ros2 node list` was empty before starting `dsr launch`

Kindly review.

leeminju531, 2024-12-05:
@gautham-nf , 
Thank you for your insights; we are looking into them.

First, when ros2 node list was empty, could you also run the following command:
`sudo grep -E libdsr_controller2.so /proc/*/maps`? This might also return empty, but it's worth checking.

Additionally, could you provide the full logs or controller logs by running this command?
`grep -R -E "ros2_control_node" ~/.ros/log/`
This will help us understand the issue further.

Lastly, could you share the steps or the launch file you used to reproduce the issue? I tested on my host by running our launch file with controllers several times, but I couldn’t reproduce the problem on my end.

gautham-nf, 2024-12-05:
As you mentioned, the command `sudo grep -E libdsr_controller2.so /proc/*/maps` gave out empty. So I think thats as per your expectation. 

Here is the ros2_control logs.. 
[ros2_control_node_logs.txt](https://github.com/user-attachments/files/18019560/ros2_control_node_logs.txt)

Steps I take to reproduce: 
- Run `ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=real host:=192.168.1.140 model:=a0509 color:=blue`
- Wait for the process to load up completely. 
- Running my launch file that would take up significant CPU(I have 2 core 4 thread CPU)
- Once the CPU hits 99-100%, the error happens. 

I tried to run the same steps in computer with better CPU. And I do not face this issue on that. 

PS: Since I understood this is a CPU issue. I can okay to close this issue and get a better PC for the cobot workstation.

closed 2025-02-24 by doosan-robotics