DoosanRobotics/doosan-robot2 #109: Wrong Values for GetCurrentRotm service call using ROS2

SakethTecholution, 2024-10-21:
`saketh@saketh-Legion-Slim-7-16IRH8:~/camera_cal$ ros2 service call /dsr01/aux_control/get_current_posx dsr_msgs2/srv/GetCurrentPosx "ref: 0" 
waiting for service to become available...
requester: making request: dsr_msgs2.srv.GetCurrentPosx_Request(ref=0)

response:
dsr_msgs2.srv.GetCurrentPosx_Response(task_pos_info=[std_msgs.msg.Float64MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[-390.9466857910156, -154.82888793945312, 113.21481323242188, 89.90401458740234, -89.98841857910156, -179.06837463378906, 2.0])], success=True)

saketh@saketh-Legion-Slim-7-16IRH8:~/camera_cal$ ros2 service call /dsr01/aux_control/get_current_rotm dsr_msgs2/srv/GetCurrentRotm "ref: 0" 
waiting for service to become available...
requester: making request: dsr_msgs2.srv.GetCurrentRotm_Request(ref=0)

response:
dsr_msgs2.srv.GetCurrentRotm_Response(rot_matrix=[std_msgs.msg.Float64MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[0.0, 0.0, 0.0]), std_msgs.msg.Float64MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[0.0, 0.0, 0.0]), std_msgs.msg.Float64MultiArray(layout=std_msgs.msg.MultiArrayLayout(dim=[], data_offset=0), data=[0.0, 0.0, 0.0])], success=True)

`
From above we can check that robot is having valid euler angles though , stil rotm service call is giving 0 values

SakethTecholution, 2024-11-01:
cc 
@doosan-robotics

leeminju531, 2024-12-18:
@SakethTecholution 
Thanks for reporting. Maybe fixed at https://github.com/doosan-robotics/doosan-robot2/pull/133

doosan-robotics, 2025-02-24:
@SakethTecholution 

This issue will be automatically closed as there has been no response from the issue reporter for over 30 days.
If you have any additional inquiries, please create a new issue.
Thank you.

closed 2025-02-24 by doosan-robotics