DoosanRobotics/doosan-robot2 #110: README installation instructions lead to colcon build errors

Cattebeke, 2024-11-04:
Hi,

I tried to install the Doosan Robot ROS2 Package on a fresh/new Ubuntu 22.04 machine, and I get errors at the penultimate step: ``` colcon build ```. See my output below. I had it working a couple of months ago on my old laptop using the same steps, but now running the ```ros2 launch ...```  - command does not work as it should.

Kind regards,
Michiel

```
~/ros2_ws$ colcon build
Starting >>> dsr_msgs2
Starting >>> ign_ros2_control
Starting >>> common2
Starting >>> dsr_description2
Starting >>> a0509_moveit_config
Starting >>> a0912_moveit_config
Starting >>> dsr_tests
Starting >>> e0509_moveit_config
Starting >>> h2017_moveit_config
Starting >>> h2515_moveit_config
Starting >>> m0609_moveit_config
Starting >>> m0617_moveit_config
Starting >>> m1013_moveit_config
Starting >>> m1509_moveit_config
Starting >>> visualservoing
Finished <<< a0509_moveit_config [0.70s]                                                                                                                                                      
Finished <<< common2 [0.74s]
Finished <<< a0912_moveit_config [0.74s]
Starting >>> example
Finished <<< m1509_moveit_config [0.97s]                                                                                                                                                  
Finished <<< dsr_description2 [1.02s]
Finished <<< m0617_moveit_config [1.00s]
Finished <<< m0609_moveit_config [1.01s]
Finished <<< e0509_moveit_config [1.02s]
Finished <<< h2515_moveit_config [1.02s]
Starting >>> dsr_gazebo2
Finished <<< m1013_moveit_config [1.02s]
Finished <<< h2017_moveit_config [1.05s]
Finished <<< dsr_tests [1.10s]                                                                                                                                                     
Finished <<< visualservoing [1.10s]
Finished <<< example [0.57s]                                                                                                            
Finished <<< dsr_gazebo2 [1.12s]                                                                                             
Finished <<< ign_ros2_control [9.91s]                                                                
Starting >>> gz_ros2_control_tests
Starting >>> ign_ros2_control_demos
Finished <<< ign_ros2_control_demos [5.95s]                                                                                                           
Finished <<< gz_ros2_control_tests [6.42s]                                                                  
[Processing: dsr_msgs2]                               
Finished <<< dsr_msgs2 [59.7s]                         
Starting >>> dsr_hardware2
Starting >>> dsr_bringup2
Starting >>> realtime_control
Finished <<< dsr_bringup2 [0.55s]                                                                                          
--- stderr: realtime_control                                                                                  
In file included from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/../../common2/include/../../common2/include/DRFL.h:58,
                 from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/../../common2/include/../../common2/include/DRFLEx.h:58,
                 from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/include/dsr_realtime_control/realtime_control.hpp:18,
                 from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/src/realtime_control.cpp:4:
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/../../common2/include/../../common2/include/DRFS.h:2205:50: warning: extra ‘;’ [-Wpedantic]
 2205 |     CONFIG_WELDING_DETAIL_INFO  _tWeldingCurrent;;
      |                                                  ^
      |                                                  -
In file included from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/../../common2/include/../../common2/include/DRFL.h:58,
                 from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/../../common2/include/../../common2/include/DRFLEx.h:58,
                 from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/include/dsr_realtime_control/realtime_api.hpp:21,
                 from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/src/realtime_api.cpp:4:
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_example2/realtime_control/../../common2/include/../../common2/include/DRFS.h:2205:50: warning: extra ‘;’ [-Wpedantic]
 2205 |     CONFIG_WELDING_DETAIL_INFO  _tWeldingCurrent;;
      |                                                  ^
      |                                                  -
---
[... 89 lines cut ...]
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp:1778:96: warning: unused parameter ‘req’ [-Wunused-parameter]
 1778 | auto stop_rt_control_cb = [this](const std::shared_ptr<dsr_msgs2::srv::StopRtControl::Request> req,
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp: In lambda function:
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp:1812:90: warning: unused parameter ‘req’ [-Wunused-parameter]
 1812 | auto read_data_rt_cb = [this](const std::shared_ptr<dsr_msgs2::srv::ReadDataRt::Request> req,
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp: In member function ‘virtual controller_interface::CallbackReturn dsr_controller2::RobotController::on_activate(const rclcpp_lifecycle::State&)’:
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp:2205:9: warning: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘struct DR_STATE’ with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
 2205 |   memset(&g_stDrState, 0x00, sizeof(DR_STATE));
      |   ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp:10:
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/include/dsr_controller2/dsr_controller2.hpp:388:16: note: ‘struct DR_STATE’ declared here
  388 | typedef struct {
      |                ^
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp: In member function ‘virtual controller_interface::return_type dsr_controller2::RobotController::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/Michiel.Cattebeke/ros2_ws/src/doosan-robot2/dsr_controller2/src/dsr_controller2.cpp:2247:24: warning: unused parameter ‘time’ [-Wunused-parameter]
 2247 |   const rclcpp::Time & time, const rclcpp::Duration & /*period*/)
      |   ~~~~~~~~~~~~~~~~~~~~~^~~~
---
Finished <<< dsr_controller2 [1min 18s]

Summary: 23 packages finished [2min 27s]
  3 packages had stderr output: dsr_controller2 dsr_hardware2 realtime_control
```

song-ms, 2024-12-16:
@Cattebeke ,

Thank you for your inquiry. The message you encountered appears to be a warning rather than an error, which means the build process should still complete successfully. You can continue to move to next step after colcon build.
If you have any further concerns or encounter unexpected behavior, please feel free to let us know, and we will assist you accordingly.

doosan-robotics, 2025-02-19:
Hello @Cattebeke 

If there are any further issues, please let me know.
I will proceed with closing this issue.

closed 2025-02-19 by doosan-robotics