DoosanRobotics/doosan-robot2 #117: ROS 2 servoj not moving the robot

KrzysztofDubel, 2024-11-21:
I have got a Doosan E0509.

I managed to have the robot move with our whole stack including ROS2 Control, Moveit2 and more, recently.

Last week I was not able to move the robot with the `servoj` function that previously worked perfectly for our use case. I've tried a few ways of doing it and I am not sure what the problem is. 

The issue was precisely that whenever the subsequent calls to `servoj` were performed, nothing happened. I updated the robot firmware to version v2.11.2.3, the `servoj` function causes the robot to violently stutter with the movement and eventually do an emergency stop. I will record some videos this week to help you understand and advise on the issue.
The `movej` function works without problems but isn't suitable for working with ROS 2 Control and Moveit planners.

Here is an extract from the code I am running which outlines all the Drfl functions I invoke and in which order they happen. (it is not the whole thing, just the doosan API functions to isolate the issue):

```cpp
CDRFLEx Drfl;

//--- doosan API's call-back fuctions : Only work within 50msec in call-back functions
Drfl.set_on_tp_initializing_completed(DSRInterface::OnTpInitializingCompletedCB);
Drfl.set_on_homming_completed(DSRInterface::OnHommingCompletedCB);
Drfl.set_on_program_stopped(DSRInterface::OnProgramStoppedCB);
Drfl.set_on_monitoring_modbus(DSRInterface::OnMonitoringModbusCB);
Drfl.set_on_monitoring_data(DSRInterface::OnMonitoringDataCB);      // Callback function in M2.4 and earlier
Drfl.set_on_monitoring_ctrl_io(DSRInterface::OnMonitoringCtrlIOCB); // Callback function in M2.4 and earlier
Drfl.set_on_monitoring_state(DSRInterface::OnMonitoringStateCB);
Drfl.set_on_monitoring_access_control(DSRInterface::OnMonitoringAccessControlCB);
Drfl.set_on_log_alarm(DSRInterface::OnLogAlarm);

if (Drfl.open_connection(m_host, m_port))
{
    while ((Drfl.GetRobotState() != STATE_STANDBY)){
            sleep(1.0);
        }

        //--- Set Robot mode : MANUAL or AUTO
        assert(Drfl.SetRobotMode(ROBOT_MODE_AUTONOMOUS));

        //--- Set Robot mode : virual or real 
        ROBOT_SYSTEM eTargetSystem = ROBOT_SYSTEM_REAL;
        assert(Drfl.SetRobotSystem(eTargetSystem));

        LPROBOT_POSE pose = Drfl.GetCurrentPose();
        
        Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_STOP);
}

while(true)
{
    Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE);
    std::array<float, NUM_JOINT> fTargetPos;
    std::array<float, NUM_JOINT> fTargetVel;
    std::array<float, NUM_JOINT> fTargetAcc;
    fTargetPos = {0.0f, 0.0f, 0.0f, 0.0f, 0.0f, 0.0f};
    fTargetVel = {120.0f, 120.0f, 149.0f, 225.0f, 225.0f, 225.0f};
    fTargetAcc = {42.0f, 38.0f, 42.0f, 80.0f, 80.0f, 120.0f};
    Drfl.servoj(fTargetPos.data(), fTargetVel.data(), fTargetAcc.data(), period.seconds());

    Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_STOP);

    sleep(period.seconds());
}

LPROBOT_POSE pose = Drfl.get_current_posj();
LPROBOT_VEL vel = Drfl.get_current_velj();

Drfl.close_connection();

void DSRInterface::OnTpInitializingCompletedCB()
{
    // request control authority after TP initialized
    cout << "[callback OnTpInitializingCompletedCB] tp initializing completed" << endl;
    g_bTpInitailizingComplted = TRUE;
    //Drfl.manage_access_control(MANAGE_ACCESS_CONTROL_REQUEST);
    Drfl.manage_access_control(MANAGE_ACCESS_CONTROL_FORCE_REQUEST);

    g_stDrState.bTpInitialized = TRUE;
}
```

Am I doing anything wrong in this extract? How could I debug why `servoj` doesn't work while `movej` does?

SakethTecholution, 2024-11-26:
I had even tried this servoj and servol directly with ROS topics . It is going to emergency stop even for me. I think it's core controller management issue 

Cc
@doosan-robotics

KrzysztofDubel, 2024-11-26:
I have managed to (sort of) fix the issue. There are a few steps / considerations:

1. The robot controller firmware needs to be updated. I updated mine to `v2.12`
2. `SAFETY_MODE_EVENT_MOVE` always has to be set before calling `servoj`
3. The max velocity and acceleration values have to be set correctly (at most slightly less than the hardware max)
4. In case of **not** using `ROS 2 Control` the `servoj` calls have to be properly coordinated, not overlap and the `time` argument has to be set properly for the movement.
5. In case of using `ROS 2 Control`
- the `time=` argument of `servoj` has to be longer than the control loop (I used times 2) 
- the initial movement has to happen in the `configure` step, **not** the `activate` step.

So the main issue with `servoj` in ROS2 (and probably the C++ API) is synchronization between the method calls and that hidden step of setting the `safety_mode`

SakethTecholution, 2024-11-26:
@KrzysztofDubel 
I am using A0509 robot . do you with the update of `v2.12` I could be able to run `MOVEIT2 `
currently i could be able to connect to the robot using `dsr_control2 pkg `. But its not updating the states in moveit2 . can you suggest what things i need to change to be able to stream moveit servo data to controller

SakethTecholution, 2024-11-26:
@KrzysztofDubel 
I am using ros2 topic of servol_stream and directly publishing the data from terminal . But its going to estop just like you  mentioned in this issue

KrzysztofDubel, 2024-11-26:
> @KrzysztofDubel
>
> I am using A0509 robot . do you with the update of `v2.12` I could be able to run `MOVEIT2 `
>
> currently i could be able to connect to the robot using `dsr_control2 pkg `. But its [...]

To make it work with moveit you'd have to modify or rewrite the hardware interface. The hardware interface in dsr_hardware2 isn't written in a way that works with moveit and ros2 control. It doesn't implement the read and write functions in a compatible way. Instead it works as a server for the services that dsr_controller2 is exposing.

So the answer is: no, not without significant changes

KrzysztofDubel, 2024-11-26:
> @KrzysztofDubel 
> 
> I am using ros2 topic of servol_stream and directly publishing the data from terminal . But its going to estop just like you  mentioned in this issue
> 
> 

Try calling the 'set_safety_stop_mode' service first and set it to MOVE (should be value 1 I think, check in the srv description). Only then use the servo streams.

Koushikraj-2005, 2024-11-27:
@KrzysztofDubel I've updated the robot to version 2.12.
I've been able to connect to the robot, but couldn't move any of the joints with the command provided in the github page.

SakethTecholution, 2024-11-27:
@Koushikraj-2005 
thanks for lettimg me know, I tried that and it worked for servol_stream but not working for servoj_stream. By any chance do you know what could be issue

I am using this commands 
`rosservice call /dsr01a0509/system/set_safety_mode "safety_mode: 1
safety_mode_event: 1" 

rostopic pub /dsr01a0509/servol_stream dsr_msgs/ServoLStream "pos: [-100.0, -500.0, 440.0, 90.0, 179.0, -175.0]
vel: [100.0, 50.0]
acc: [100.0, 50.0]
time: 0.0" 

rostopic pub /dsr01a0509/servoj_stream dsr_msgs/ServoJStream "pos: [270.0, 0.0, 90.0, 0.0, 90.0, 0.0]
vel: [50.0, 50.0, 50.0, 50.0, 50.0, 50.0]
acc: [50.0, 50.0, 50.0, 50.0, 50.0, 50.0]
time: 10.0" 
`

SakethTecholution, 2024-11-27:
@Koushikraj-2005 
Yeah everyone is actually facing the same issue and doosan team isn't responding to the cause too . As @KrzysztofDubel said it's the core dsr hardware interface package itself is an issue and if u lookedinto moveit pkg it's hard-coded to m1013. I think they haven't tested moveit functionality yet on robot.

closed 2025-04-09 by doosan-robotics