DoosanRobotics/doosan-robot2 #119: ROS 2 Humble  - Troubles with configuration of A0509

DRAJesse, 2024-11-25:
I'm reaching out for some assistance with configuring the A0509 model in ROS 2 Humble. I've cloned the doosan-robotics/doosan-robot2 repository from GitHub, specifically the humble-devel branch, and I'm running it on an x86 system with Ubuntu. While ROS 2 Humble itself runs perfectly on my system, I'm encountering issues when working with the A0509 model.

 

Initially, when I launched the launch file, I faced the following error: [rviz2-6] Error: Semantic description is not specified for the same robot as the URDF [rviz2-6] at line 681 in ./src/model.cpp

 I found a mismatch in the robot names; the SRDF file used 'dsr' while the URDF file named the robot 'a0509'. After correcting this inconsistency, I am still facing persistent errors both in the terminal and RViz.

Could you possibly provide me with a working example or any guidance on setting up the A0509 model using ROS 2 Humble? I've also attached screenshots of the terminal and the RViz.

 

Any assistance you can provide would be greatly appreciated as I work through these configuration challenges.

 
[image]

[image]
[image]
[image]

SakethTecholution, 2024-11-26:
We are even facing the same issue . Currently I don't think the doosan is giving support for using moveit on ros 2 . They harcoded the robot name to m1013 . Moveit functionality is not working with ROS2

leeminju531, 2024-12-09:
Dear @DRAJesse and @SakethTecholution,

We are currently investigating the issue you're facing with MoveIt. The bug appears to occur during the MoveIt launch process. Our team is actively working on resolving the problem, and we will provide an update as soon as possible.

Thank you for your patience, and we will notify you once the issue has been addressed.

leeminju531, 2024-12-17:
Dear @DRAJesse @SakethTecholution 

The related feature has been fixed in [this PR](https://github.com/doosan-robotics/doosan-robot2/pull/125). Could you please try it again? Feel free to reach out if you encounter any issues.

doosan-robotics, 2025-02-24:
@DRAJesse 

This issue will be automatically closed as there has been no response from the issue reporter for over 30 days.
If you have any additional inquiries, please create a new issue.
Thank you.

closed 2025-02-24 by doosan-robotics