DoosanRobotics/doosan-robot2 #126: Moveit2 with A0509

SakethTecholution, 2024-12-13:
@leeminju531 @doosan-robotics 
Can you please let us know how to use A0509 robot with moveit2 because we are heavily dependent on moveit2 but currently it's not working. Can you please let us know atleast the timeline or possible issues to make it work

song-ms, 2024-12-16:
@SakethTecholution 

I apologies for the delayed response. I have updated the features related to MoveIt2.
Please check bringup moveit2 again.

gautham-nf, 2024-12-24:
@SakethTecholution: Does moveit work for you with the actual robot? 

For me currently moveit works in RVIZ. It plans path, but it does not get executed on the robot. 

[image]

It even logs all is working well. But the robot does not move.

gautham-nf, 2024-12-27:
@song-ms @leeminju531. Could you kindly help me with this issue.

leeminju531, 2024-12-27:
Dear @gautham-nf 

First, could you try deleting the build and install folders, then rebuild the packages ? 

If the issue persists, please open a new issue and include the full logs.

gautham-nf, 2024-12-27:
Thanks for the response

I did try to rebuild the whole package after deleting. 
In the logs I could not see any any problem. 

I shall create a new issue.

doosan-robotics, 2025-02-19:
If there are any further issues, please provide an update.
Otherwise, this issue will be closed.

closed 2025-02-19 by doosan-robotics