DoosanRobotics/doosan-robot2 #127: Publishing JointStates to Move robot

SakethTecholution, 2024-12-13:
We are using external motion planer to move A0509 robot. But we need to send joint commands continuously to robot. I have been looking into servoJ_stream service to send it but it isn't working. I tried movej service with async call but since the frequency of hitting is high it's going to save off .
In this case can you recommend what is the better way to send joint commands continuously to doosan robot 
@leeminju531 @doosan-robotics

SakethTecholution, 2024-12-16:
@song-ms

song-ms, 2024-12-16:
@SakethTecholution 

Thank you for your inquiry. We mostly recommand servoj_stream or blending motion for continuously command motion.
Could you clarify what specific issues you encountered while using the servoJ_stream service? Additionally, it would be helpful to know what commands you were sending at the time and how they were structured. This information will allow us to better understand the problem and recommend an appropriate solution.

doosan-robotics, 2025-02-19:
If there are any further issues, please provide an update.
Otherwise, this issue will be closed.

closed 2025-02-19 by doosan-robotics