DoosanRobotics/doosan-robot2 #128: MoveIt error "Requesting initial scene failed"

airizar98, 2024-12-13:
We are working with an H2017 robot. When we launch "dsr_bringup2_moveit", it connects correctly to the robot, but a warning appears in RVIZ under PlanningScene and MotionPlanning, showing the following message: "Requesting initial scene failed." The issue is that it does not receive the robot's current position; in RVIZ, all axes appear at 0º. When we move the axes in RVIZ and try to plan and execute, it doesn’t work.
However, when we launch the command "dsr_bringup2_rviz", the robot's position is displayed correctly. In this case, though, since the launch does not execute MoveIt, we get an error when trying to add MotionPlanning.

song-ms, 2024-12-16:
@airizar98 ,

Thank you for your inquiry. We have newly updated the features related to MoveIt. Please ensure that the controller version is set to v2.12 or v2.12.1, and update accordingly before use.

airizar98, 2024-12-19:
Thank you for the response. We updated the robot’s version and the ROS2 repository, and now MoveIt successfully plans paths. However, when we press "Execute" on the real robot (inside RVIZ), the robot starts making strange noises and barely moves. Errors related to dsr_hw_interface2 appear in the command line, specifically:
“Target limit velocity limit error, [2] joint, target limit velocity [100.000 deg/s], limit velocity [80.000 deg/s].”
Also appear issues regarding acceleration limits.

We modified the h2017 joint_limits.yaml by setting all velocity and acceleration limits to false, but the issue persists.
[image]

leeminju531, 2024-12-20:
@airizar98 

Could we know your controller update rate in `dsr_contrller2/config/dsr_controller2.yaml`? 

```/**:
  controller_manager:
    ros__parameters:
      update_rate: 10  # Hz
```

airizar98, 2024-12-20:
We have the same value:
[image]

song-ms, 2024-12-26:
@airizar98 

Could you please delete all the build install logs in your workspace before building and try building again?

Also, could you let me know if the LED is in blinking mode in auto mode(white) during "execute" after the robot boots up?

leeminju531, 2024-12-27:
Dear @airizar98 
If the issue still persists, could you please try it on https://github.com/doosan-robotics/doosan-robot2/pull/140 and let us know the results?

airizar98, 2025-01-07:
Dear @leeminju531 ,
The robot works correctly with the latest updates. However, it is unable to load the lates h2017 URDF version. I had to revert to the previous version, even though it is not completely correct.
Thank you very much for the help.

doosan-robotics, 2025-02-19:
If there are any further issues, please provide an update.
Otherwise, this issue will be closed.

closed 2025-02-19 by doosan-robotics