DoosanRobotics/doosan-robot2 #137: ros2 humble M1013 질문입니다.

kwongeunwoo, 2024-12-20:
안녕하세요 ROS2 Humble에서 doosan-robot2 pkg를 이용하여 M1013를 제어하고자 하는 사용자입니다.

다름이 아니라 로봇에 joint input을 velocity로 velocity control을 원하는데, example 코드를 참고하여 제어하려 했으나 common2/imp/DSR_ROBOT2.py에는 speedj와 같은 velocity control cmd 함수가 없어 혹시 ros2 humble에서 속도 제어를 할 수 있는 방법이 있는지 질문드립니다. 

읽어주셔서 감사합니다!

rundohyeon, 2024-12-20:
set_velx, set_accx, set_velj, set_accj 로 속도 제어 가능합니다

kwongeunwoo, 2024-12-20:
답변 감사합니다. 하지만 제가 궁금한 점은 joint desired pose없이 velocity를 input으로 바로 넣을 수 있는 함수가 있는지가 궁금합니다. 예를들어 1번 모터가 속도 -10 rad/s로 돌아라 이런식의 input으로 제어가 가능한 함수를 사용 가능한지 궁금합니다.

rundohyeon, 2024-12-20:
servoing 말씀이신것같네요
써본적은 없지만 이 example처럼 목표 위치를 현재위치 대비하여 계속 바꿔주는 방법이 있지 않을까 싶습니다

https://github.com/doosan-robotics/doosan-robot2/tree/humble-devel/dsr_example2/visualservoing

rundohyeon, 2024-12-20:
지금 찾아보니 jog라는 함수가 있네요
```

def jog(jog_axis, ref=0, speed=0):
    if type(jog_axis) != int and type(jog_axis) != float:
        raise DR_Error(DR_ERROR_TYPE, "Invalid type : jog_axis")
    
    if type(ref) != int:
        raise DR_Error(DR_ERROR_TYPE, "Invalid type : ref")

    if type(speed) != int and type(speed) != float:
        raise DR_Error(DR_ERROR_TYPE, "Invalid type : speed")
    # ROS service call
    if __ROS2__:
        req = Jog.Request()
        req.jog_axis       = jog_axis                            
        req.move_reference = ref
        req.speed          = float(speed)

        future = _ros2_jog.call_async(req)
        rclpy.spin_until_future_complete(g_node, future)

        try:
            result = future.result()
        except Exception as e:
            g_node.get_logger().info('jog Service call failed %r' % (e,))
        else:
            if result == None:
                ret = -1    
            else:        
                ret = 0 if (result.success == True) else -1     
    return ret

```

kwongeunwoo, 2024-12-21:
답변 감사합니다. 한번 시도해보겠습니다!

song-ms, 2025-01-22:
@kwongeunwoo 님

안녕하세요 해당 이슈가 완료 되었다면 이슈 클로즈 해도 괜찮을까요 ?

kwongeunwoo, 2025-01-23:
넵!

closed 2025-01-24 by doosan-robotics