DoosanRobotics/doosan-robot2 #142: Services Unavailable in MoveIt

Manimaran017, 2024-12-30:
Description:
The program script works in RViz but the services (Set Robot Mode Service and MoveJ Service) are unavailable when running with MoveIt. The script hangs, waiting for the services to become available, and eventually fails.

Steps I Followed:
1. Build the workspace using colcon build 
2. source install/setup.bash
3. ros2 launch dsr_bringup2 dsr_bringup2_moveit.launch.py mode:=real model:=a0912 host:=192.168.137.100
4. ros2 run example simple

Expected Behavior:
The script should initialize the required services and work with MoveIt to execute robot motions.

Actual Behavior:
The script works correctly in RViz.
When used with MoveIt, it waits indefinitely for the services (Set Robot Mode Service and MoveJ Service) and fails to execute.

Request:
Please investigate why the services are not available with MoveIt, even though the script works in RViz. Any guidance or fixes would be greatly appreciated
[image]
[image]

leeminju531, 2024-12-30:
Thank you for your inquiry
It seems that when launching MoveIt, there is no default namespace. However, in the example file you provided, it appears the namespace is set to 'dsr01'.
Please feel free to reach out if the issues persist!

Manimaran017, 2024-12-30:
Hi,

I have followed the instructions to set up my robot with MoveIt, but I am encountering some issues. After launching the robot and running the example, I see the following logs:

It appears that the "Services" are not available, and there are errors related to a mismatch between the URDF and SRDF files.

Could you please assist in resolving these issues, specifically the URDF/SRDF mismatch and the unavailable service?

Steps I have taken:
1)Verified that the namespace (dsr01) is set correctly.
2)Ensured that the robot model (a0912) is specified in the launch file.
3)The "Set Robot Mode" service is still not available after waiting for some time. Please see the attached screenshot below for further context.
[image]

Thank you for your assistance. I look forward to your support in resolving this issue.

Manimaran017, 2025-01-03:
Hi @leeminju531 
I wanted to kindly follow up on this issue.

leeminju531, 2025-01-03:
Dear @Manimaran017,

Could you make service calls (like SetRobotMode and movejoint) without running the example file through the CLI?
If it works, please describe the example file we could run. If not, please provide what you did and the full log.

Manimaran017, 2025-01-04:
Hi @leeminju531,
It's not working. I attached the log files

[service commands.txt](https://github.com/user-attachments/files/18305698/service.commands.txt)
[MoveIt command.txt](https://github.com/user-attachments/files/18305699/MoveIt.command.txt)

Manimaran017, 2025-01-06:
Dear @leeminju531,

Just following up on the issue I reported earlier. Have there been any updates on the service call or potential solutions for the errors?

leeminju531, 2025-01-07:
@AstrobotDaniel 

> 1. The service in question seems to be provided via dsr_bringup2_gazebo.launch.py

This issue maybe related to `dsr_bringup2_moveit.launch.py `

> 2. dsr_bringup2_moveit.launch.py requires namespace:=dsr01 to work (this is missing in the README.md)
a. With the proper namespace defined, RVIZ launches with Moveit enabled, otherwise it fails to start-up

After further investigation, I found that `dsr_bringup2_moveit.launch.py` doesn't work when a namespace is set. 

> b. While the move_joint service can be called with this setup, it results in faulty response initially (success=False):

This could be due to the controller still being in the boot-up process, not fully initailized yet.

> c. if the motion planner was used once, the gazebo model responds to the service call, but the rviz robot remains where it is:

This is caused by the namespace issue.

I will create a follow-up issue to add namespace support in `dsr_bringup2_moveit.launch.py`.
In the meantime, could you try running it without specifying the namespace?
(e.g.: `ros2 launch dsr_bringup2 dsr_bringup2_moveit.launch.py mode:=virtual host:=127.0.0.1`.)

leeminju531, 2025-01-07:
I believe the initial failure of the `move_joint` service and the `dsr_bringup2_moveit.launch.py` failure (due to the connection error) addressed in https://github.com/doosan-robotics/doosan-robot2/pull/144

AstrobotDaniel, 2025-01-07:
I updated to https://github.com/doosan-robotics/doosan-robot2/commit/1808c11ba05a452f5fec3224313793d281154d9e in order to test your solution merged with #144. Unfortunately, this makes things worse.

Fist of all, the problem with the missing service remains. If I just start `dsr_bringup2_moveit.launch.py` without specifying a namespace and then run `ros2 service call /motion/move_joint dsr_msgs2/srv/MoveJoint ..` the call stays at `waiting for service to become available...` indefinitely. Am I supposed to start `dsr_bringup2_gazebo.launch.py` in addition?

Yet even worse, the update seams to break the controller. If I use the RViz GUI to call Plan & Execute, the robot is not following the plan and the service returns with a timeout. 

For comparison, I initiated the planner through the GUI twice, once with the new version https://github.com/doosan-robotics/doosan-robot2/commit/1808c11ba05a452f5fec3224313793d281154d9e and once with the previous working version https://github.com/doosan-robotics/doosan-robot2/commit/0460ba72ef8e6860d602dbf24c85035a32c1fc5c. Both output files are attached here for your review.

new failing version: [1808c11.txt](https://github.com/user-attachments/files/18337959/1808c11.txt)
old working version: [0460ba7.txt](https://github.com/user-attachments/files/18337960/0460ba7.txt)

Manimaran017, 2025-01-08:
@leeminju531 
I’m encountering similar issues with the controller not following the planned trajectory and the service call timing out. Both real-time and virtual modes seem to be affected with RViz working in real-time but not in virtual mode and MoveIt not working in either mode. Let me know if you find a solution and I’ll continue testing on my end as well.

leeminju531, 2025-01-09:
@AstrobotDaniel 
> Am I supposed to start dsr_bringup2_gazebo.launch.py in addition?

No, you only need to start `dsr_bringup2_moveit.launch.py.`

@AstrobotDaniel @Manimaran017 
The PR I sent has been reverted—sorry for the inconvenience and thanks for your investigations.

I believe the issues occurred due to two main points:

1. The controller needs to be fully invoked, which is related to the two issues mentioned:
- Moveit timeout failure (100 % related)
`[ros2_control_node-2] [ERROR] [1735965227.344591286] [dsr_hw_interface2]: DSRInterface::init() DRCF connecting ERROR!!!`
- move_joint service call related (might be related)

2. `dsr_bringup2_moveit.launch.py` needs to support namespace arguments.

I’ll open separate follow-up issues for this. Thanks again!

Manimaran017, 2025-01-10:
Hi @leeminju531 ,

Thank you for your update regarding the issues you’ve identified. I appreciate your thorough investigation and your plan to open separate follow-up issues.

I would like to revisit the URDF-related warnings that we discussed earlier. The warnings indicate the following:

1) Namespace Collision Warning: 

[rviz2-8] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-8]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-8] Error:   Semantic description is not specified for the same robot as the URDF
[rviz2-8]          at line 664 in ./src/model.cpp

This warning suggests a potential issue with library linking that could complicate our setup. This might need addressing to ensure that all robot descriptions are properly aligned.

Additionally, I have a concern regarding the virtual environment. I’m uncertain if I need to run a program in a virtual setup to verify how the robot moves. However, when I do run a program in the virtual environment, I don’t see any movement in RViz or Gazebo, although the services do seem to function correctly. If you have any insights into this issue(how run a program in virtual mode), it would be greatly appreciated.

While you mentioned that the initial warnings do not currently affect the services, I believe it is still important to resolve them to prevent any future complications, especially as we work towards a stable integration.

Thank you again for your efforts and I look forward to your updates on the follow-up issues.

leeminju531, 2025-01-10:
@Manimaran017, 

> While you mentioned that the initial warnings do not currently affect the services, I believe it is still important to resolve them to prevent any future complications, especially as we work towards [...]

I totally agree. I’ve also opened a separate URDF-related issue as well.

> Additionally, I have a concern regarding the virtual environment. I’m uncertain if I need to run a program in a virtual setup to verify how the robot moves. However, when I do run a program in the virtual environment, I don’t see any movement in RViz or Gazebo, although the services do seem to function correctly. If you have any insights into how to run a program in virtual mode, it would be greatly appreciated.

Regarding your concern with the virtual environment, I think you’re approaching it the right way.
However, it seems like it’s not working due to a lack of resources. If you're running it on a virtual machine, that could be the issue. 
In my opinion, at the very least, it should provide a notification indicating the cause, even with resource limitations. 
Unfortunately, I’m not sure I can address this in a follow-up issue since I can't diagnose the exact reason at the moment.
For now, I’ll focus on investigating the `Moveit timeout failure `issue, which might be somewhat related to this. Once get merged, I’ll let you know.

leeminju531, 2025-02-04:
@Manimaran017 
Apologies for the late notice. The Moveit timeout failure might be addressed in #151 first. Thanks!

closed 2025-02-19 by doosan-robotics