DoosanRobotics/doosan-robot2 #152: spawner_joint_state_broadcaster 및 spawner_dsr_controller2 이슈

hanprd, 2025-01-16:
안녕하세요.
Ubuntu 22.04.5 LTS에서 ROS2 Humble 기반의 DSR 패키지 실행시 이슈가 발생하여 문의를 드립니다.
주요 오류는 Servo on까지 정상적으로 이루어지나 아래 메시지와 함께 최종 연결까지 이루어지지 않는 것입니다.
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [connection-1]: process started with pid [1947017]
[INFO] [ros2_control_node-2]: process started with pid [1947019]
[INFO] [robot_state_publisher-3]: process started with pid [1947021]
[INFO] [spawner-4]: process started with pid [1947023]
[INFO] [spawner-5]: process started with pid [1947025]
[ros2_control_node-2] [WARN] [1737044563.354498570] [dsr01.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-2] [INFO] [1737044563.354927061] [resource_manager]: Loading hardware 'm0609' 
[ros2_control_node-2] [INFO] [1737044563.376889154] [resource_manager]: Initialize hardware 'm0609' 
[robot_state_publisher-3] [INFO] [1737044563.405445158] [dsr01.robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1737044563.405578354] [dsr01.robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1737044563.405593521] [dsr01.robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1737044563.405602868] [dsr01.robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1737044563.405611490] [dsr01.robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1737044563.405620027] [dsr01.robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1737044563.405628497] [dsr01.robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1737044563.405636802] [dsr01.robot_state_publisher]: got segment world
**[spawner-5] [INFO] [1737044563.748886169] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-4] [INFO] [1737044563.752184009] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...**
[ros2_control_node-2] [INFO] [1737044571.377415745] [dsr_hw_interface2]: [on_init] Hardware name : m0609, type : system, class type : dsr_hardware2/DRHWInterface
[ros2_control_node-2] [INFO] [1737044571.385118252] [dsr01.dsr_hw_interface2]: name: dsr01
[ros2_control_node-2] [INFO] [1737044571.385209586] [dsr01.dsr_hw_interface2]: rate: 100
[ros2_control_node-2] [INFO] [1737044571.385279014] [dsr01.dsr_hw_interface2]: standby: 5000
[ros2_control_node-2] [INFO] [1737044571.385315462] [dsr01.dsr_hw_interface2]: command: true
[ros2_control_node-2] [INFO] [1737044571.385337925] [dsr01.dsr_hw_interface2]: host: 192.168.137.100
[ros2_control_node-2] [INFO] [1737044571.385363786] [dsr01.dsr_hw_interface2]: port: 12345
[ros2_control_node-2] [INFO] [1737044571.385385479] [dsr01.dsr_hw_interface2]: mode: real
[ros2_control_node-2] [INFO] [1737044571.385428587] [dsr01.dsr_hw_interface2]: model: m0609
[ros2_control_node-2] [INFO] [1737044571.385449262] [dsr01.dsr_hw_interface2]: gripper: none
[ros2_control_node-2] [INFO] [1737044571.385490038] [dsr01.dsr_hw_interface2]: mobile: none
[ros2_control_node-2] [INFO] [1737044571.385628938] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] [INFO] [1737044571.385662100] [dsr_hw_interface2]:     INITAILIZE
[ros2_control_node-2] [INFO] [1737044571.385677089] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] Depreciated API.. deleted soon.
[ros2_control_node-2]  Please use 'set_on_monitoring_data_ex'
[ros2_control_node-2] Depreciated API.. deleted soon.
[ros2_control_node-2]  Please use 'set_on_monitoring_ctrl_io_ex'.
[ros2_control_node-2] [INFO] [1737044571.498234141] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] [INFO] [1737044571.498275980] [dsr_hw_interface2]:     OPEN CONNECTION
[ros2_control_node-2] [INFO] [1737044571.498283727] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] [INFO] [1737044571.498960850] [dsr_hw_interface2]:     Real Robot Mode
[ros2_control_node-2] [INFO] [1737044571.498974488] [dsr_hw_interface2]:     DRCF version = GF03020200
[ros2_control_node-2] [INFO] [1737044571.498985817] [dsr_hw_interface2]:     DRFL version = GL013100
[ros2_control_node-2] [INFO] [1737044571.498991720] [dsr_hw_interface2]:     m_nVersionDRCF = 3020200
[ros2_control_node-2] [INFO] [1737044571.498997373] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] [INFO] [1737044571.689094662] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] [INFO] [1737044571.689133736] [dsr_hw_interface2]:     Access control granted 
[ros2_control_node-2] [INFO] [1737044571.689140516] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2] 
[ros2_control_node-2] usClientPort : 56681
[ros2_control_node-2] [INFO] [1737044573.461145089] [dsr_hw_interface2]: [callback OnLogAlarm]
[ros2_control_node-2] [INFO] [1737044573.461173159] [dsr_hw_interface2]:  level : 1
[ros2_control_node-2] [INFO] [1737044573.461180036] [dsr_hw_interface2]:  group : 1
[ros2_control_node-2] [INFO] [1737044573.461185250] [dsr_hw_interface2]:  index : 1037
[ros2_control_node-2] [INFO] [1737044573.461190640] [dsr_hw_interface2]:  param : 
[ros2_control_node-2] [INFO] [1737044573.461195748] [dsr_hw_interface2]:  param : 
[ros2_control_node-2] [INFO] [1737044573.461200772] [dsr_hw_interface2]:  param : 
**[spawner-5] [WARN] [1737044573.763732195] [dsr01.spawner_joint_state_broadcaster]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-5] [INFO] [1737044573.764302125] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-4] [WARN] [1737044573.766772696] [dsr01.spawner_dsr_controller2]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-4] [INFO] [1737044573.767326025] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...**
[ros2_control_node-2] [WARN] [1737044578.567277267] [dsr_hw_interface2]: [callback OnLogAlarm]
[ros2_control_node-2] [WARN] [1737044578.567305660] [dsr_hw_interface2]:  level : 2
[ros2_control_node-2] [WARN] [1737044578.567312580] [dsr_hw_interface2]:  group : 1
[ros2_control_node-2] [WARN] [1737044578.567318035] [dsr_hw_interface2]:  index : 3014
[ros2_control_node-2] [WARN] [1737044578.567323483] [dsr_hw_interface2]:  param : A heartbeat packet was not received for 5 seconds
[ros2_control_node-2] [WARN] [1737044578.567328905] [dsr_hw_interface2]:  param : 
[ros2_control_node-2] [WARN] [1737044578.567334107] [dsr_hw_interface2]:  param : 
**[spawner-5] [WARN] [1737044583.776659152] [dsr01.spawner_joint_state_broadcaster]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-5] [INFO] [1737044583.777208537] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-4] [WARN] [1737044583.779316439] [dsr01.spawner_dsr_controller2]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-4] [INFO] [1737044583.779857782] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...**
ROS1 패키지 및 윈도우 기반의 Dart에서는 정상적으로 연결 및 동작이 되는 것을 확인하였습니다.
이슈 해결에 도움을 주시면 감사하겠습니다.

song-ms, 2025-01-19:
안녕하세요 @hanprd 님,

문의 주신 내용을 파악하기위해 확인차 질문을 드립니다. 
/{workspace}/src/doosan_robot2 경로에 git pull- r 으로 ros 코드가 최신 업데이트 되어있는지 확인 부탁드리며,
혹시 어떤 명령어를 사용하였을때 저런 현상이 발생하는지 알수 있을까요 ?

hanprd, 2025-01-20:
> ---- External Email: Use caution with attachments, links, or sharing data ----

안녕하세요?
사용한 명령어는 아래와 같고 최신 버전의 코드 사용하였습니다.

ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=real host:=192.168.137.100 port:=12345 model:=m0609

감사합니다.

song-ms, 2025-01-20:
@hanprd 님

확인 감사합니다. 해당 내용을 확인하고 이슈부분 수정 완료 하였습니다. 
수정된 소스 업데이트 후 재확인 부탁드립니다.

hanprd, 2025-01-20:
조치 감사드립니다.
다만 업데이트 후 가상모드 실행이 되지 않습니다.
실제 환경에 대해서는 테스트 후 말씀드리겠습니다.

Command:~/ros2_ws$ ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=virtual host:=127.0.0.1 port:=12345 model:=m0609
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2025-01-20-10-14-19-852323-pc-3362272
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [connection-1]: process started with pid [3362280]
[INFO] [ros2_control_node-2]: process started with pid [3362282]
[INFO] [robot_state_publisher-3]: process started with pid [3362284]
[INFO] [spawner-4]: process started with pid [3362287]
[INFO] [spawner-5]: process started with pid [3362290]
[ros2_control_node-2] [WARN] [1737386060.119693612] [dsr01.controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-2] [INFO] [1737386060.120026777] [resource_manager]: Loading hardware 'm0609'
[ros2_control_node-2] [INFO] [1737386060.127824817] [resource_manager]: Initialize hardware 'm0609'
[ros2_control_node-2] [INFO] [1737386060.128005583] [dsr_hw_interface2]: [on_init] Hardware name : m0609, type : system, class type : dsr_hardware2/DRHWInterface
[ros2_control_node-2] [ERROR] [1737386060.133332130] [dsr01.dsr_hw_interface2]: Failed to open YAML file: /home/user/ros2_ws/install/dsr_hardware2/share/dsr_hardware2/config/dsr01_parameters.yaml
[ros2_control_node-2] [INFO] [1737386060.133417604] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2]
[ros2_control_node-2] [INFO] [1737386060.133442333] [dsr_hw_interface2]:     INITAILIZE
[ros2_control_node-2] [INFO] [1737386060.133444565] [dsr_hw_interface2]: _______________________________________________
[ros2_control_node-2]
[robot_state_publisher-3] [INFO] [1737386060.138259767] [dsr01.robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1737386060.138348857] [dsr01.robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1737386060.138354635] [dsr01.robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1737386060.138357524] [dsr01.robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1737386060.138360611] [dsr01.robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1737386060.138363433] [dsr01.robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1737386060.138365969] [dsr01.robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1737386060.138368589] [dsr01.robot_state_publisher]: got segment world
[spawner-4] [INFO] [1737386060.555980367] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-5] [INFO] [1737386060.558369158] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[connection-1] [INFO] [1737386060.560000703] [dsr01.connection_node]: @@@@@@@@@@@@@@@@@@@@@@@@@@@ run_drcf
[connection-1] Run Emulator of the Doosan Robot Controller
[connection-1] dirname: /home/user/ros2_ws/install/common2/share/common2/bin/run_drcf.sh
[connection-1] server_port: 12345
[connection-1] robot model: m0609
[connection-1] ns: dsr01
[connection-1] run : 'dsr01_emulator' ...
[connection-1] bde436b315dfba441f7e604dffdd4e28a19186a9fa411b0634e706519f013f59
[connection-1] ARCH: 64-bit
[spawner-4] [WARN] [1737386070.569415833] [dsr01.spawner_dsr_controller2]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-4] [INFO] [1737386070.569666398] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-5] [WARN] [1737386070.571434211] [dsr01.spawner_joint_state_broadcaster]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-5] [INFO] [1737386070.571788287] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-4] [WARN] [1737386080.583305575] [dsr01.spawner_dsr_controller2]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-4] [INFO] [1737386080.583524873] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-5] [WARN] [1737386080.584757936] [dsr01.spawner_joint_state_broadcaster]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-5] [INFO] [1737386080.585158545] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-4] [WARN] [1737386090.597149414] [dsr01.spawner_dsr_controller2]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-4] [INFO] [1737386090.597389206] [dsr01.spawner_dsr_controller2]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[spawner-5] [WARN] [1737386090.597860099] [dsr01.spawner_joint_state_broadcaster]: Could not contact service /dsr01/controller_manager/list_controllers
[spawner-5] [INFO] [1737386090.598220629] [dsr01.spawner_joint_state_broadcaster]: waiting for service /dsr01/controller_manager/list_controllers to become available...
[... cut ...]
받는 사람: doosan-robotics/doosan-robot2 ***@***.***>
참조: Han, Changheon ***@***.***>; Mention ***@***.***>
제목: Re: [doosan-robotics/doosan-robot2] spawner_joint_state_broadcaster 및 spawner_dsr_controller2 이슈 (Issue #152)

***@***.***에게서 전자 메일을 받지 못하는 경우가 많습니다. 이 문제가 중요한 이유<https://aka.ms/LearnAboutSenderIdentification>
---- External Email: Use caution with attachments, links, or sharing data ----

@hanprd<https://github.com/hanprd> 님

확인 감사합니다. 해당 내용을 확인하고 이슈부분 수정 완료 하였습니다.
수정된 소스 업데이트 후 재확인 부탁드립니다.

song-ms, 2025-01-21:
@hanprd 님, 안녕하세요

virtual 관련해서 첫 연결시 이슈 연결이 안되는 현상이 있어 해당 부분을 함께 수정했습니다.
다시한번 업데이트 후 확인 부탁드립니다.

감사합니다.

hanprd, 2025-01-21:
먼저 기존의 두산 로봇 패키지를 제거하고 rosdep 초기화 및 업데이트와 함께 다른 우분투 소프트웨어도 업데이트를 한 후 개정된 패키지를 재설치 하였음에도 불구하고 spawner_joint_state_broadcaster 및 spawner_dsr_controller2 이슈가 virtual, real 모드에서 발생합니다.
3개의 시스템으로 테스트해보았고 로봇 간의 통신에는 ping 명령어를 통해 이상없음을 확인하였습니다.
지난번에는 servo on까지는 되었었는데 이번에는 이 부분도 되지 않네요.
다시 한번 확인해주시면 감사하겠습니다.

________________________________
보낸 사람: Minsoo Song ***@***.***>
보낸 날짜: 2025년 1월 21일 화요일 오전 1:19
받는 사람: doosan-robotics/doosan-robot2 ***@***.***>
참조: Han, Changheon ***@***.***>; Mention ***@***.***>
제목: Re: [doosan-robotics/doosan-robot2] spawner_joint_state_broadcaster 및 spawner_dsr_controller2 이슈 (Issue #152)

***@***.***에게서 전자 메일을 받지 못하는 경우가 많습니다. 이 문제가 중요한 이유<https://aka.ms/LearnAboutSenderIdentification>
---- External Email: Use caution with attachments, links, or sharing data ----

@hanprd<https://github.com/hanprd> 님, 안녕하세요

virtual 관련해서 첫 연결시 이슈 연결이 안되는 현상이 있어 해당 부분을 함께 수정했습니다.
다시한번 업데이트 후 확인 부탁드립니다.

감사합니다.

leeminju531, 2025-01-22:
@hanprd 님, 안녕하세요. 

관련사항 업데이트 하였습니다. 최신 커밋으로 확인 부탁드립니다.

cc @song-ms

closed 2025-01-23 by doosan-robotics