DoosanRobotics/doosan-robot2 #158: Launch With Moveit2 이슈

hanprd, 2025-01-22:
안녕하세요.
최신 커밋을 통해 L"Launch With Rviz2" 모드 실행이 정상적으로 이루어지는 것을 확인하였습니다만 
"Launch With Moveit2" 모드 실행이 되지 않아 문의를 남깁니다.
Ubuntu 22.04.5 LTS OS 아래 ROS2 Humble 환경에서 실행하였으며 로봇은  m0609입니다.
빌드는 colcon build --cmake-args -DDRCF_VER=3  명령어로 수행하였습니다.
아래에 실행 로그와 GUI 화면 첨부하였습니다.

lamm@lamm-OptiPlex-790:~$ ros2 launch dsr_bringup2 dsr_bringup2_moveit.launch.py mode:=real model:=m0609 host:=192.168.137.100
[INFO] [launch]: All log files can be found below /home/lamm/.ros/log/2025-01-22-11-38-56-799547-lamm-OptiPlex-790-1686242
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
[INFO] [set_config-1]: process started with pid [1686249]
[INFO] [run_emulator-2]: process started with pid [1686251]
[INFO] [robot_state_publisher-3]: process started with pid [1686253]
[INFO] [spawner-4]: process started with pid [1686255]
[INFO] [spawner-5]: process started with pid [1686257]
[INFO] [spawner-6]: process started with pid [1686260]
[INFO] [move_group-7]: process started with pid [1686262]
[robot_state_publisher-3] [INFO] [1737563937.404151919] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1737563937.408408070] [robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1737563937.408432318] [robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1737563937.408442199] [robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1737563937.408451169] [robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1737563937.408459765] [robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1737563937.408468357] [robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1737563937.408476775] [robot_state_publisher]: got segment world
[move_group-7] Error:   Semantic description is not specified for the same robot as the URDF
[move_group-7]          at line 681 in ./src/model.cpp
[move_group-7] [INFO] [1737563937.516241478] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0914237 seconds
[move_group-7] [INFO] [1737563937.516315373] [moveit_robot_model.robot_model]: Loading robot model 'm0609'...
[run_emulator-2] Run Emulator of the Doosan Robot Controller
[run_emulator-2] dirname: /home/lamm/ros2_ws/install/common2/share/common2/bin/run_drcf.sh
[run_emulator-2] server_port: 12345
[run_emulator-2] robot model: m0609
[run_emulator-2] ns: 
[run_emulator-2] run : 'emulator' ...
[move_group-7] [WARN] [1737563937.691602821] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[spawner-4] [INFO] [1737563937.873422224] [spawner_dsr_controller2]: waiting for service /controller_manager/list_controllers to become available...
[spawner-6] [INFO] [1737563937.984344643] [spawner_dsr_moveit_controller]: waiting for service /controller_manager/list_controllers to become available...
[move_group-7] [INFO] [1737563937.998073046] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-7] [INFO] [1737563937.998230966] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-7] [INFO] [1737563937.999628478] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-7] [INFO] [1737563937.999987160] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-7] [INFO] [1737563938.000004433] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-7] [INFO] [1737563938.000210028] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-7] [INFO] [1737563938.000224510] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-7] [INFO] [1737563938.000488489] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-7] [INFO] [1737563938.000744360] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-7] [WARN] [1737563938.002463525] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-7] [ERROR] [1737563938.002489232] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[spawner-5] [INFO] [1737563938.013020120] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[move_group-7] [INFO] [1737563938.231186117] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[move_group-7] [ERROR] [1737563938.232524707] [moveit.ros_planning.planning_pipeline]: Exception while loading planner 'ompl_interface/OMPLPlanner': According to the loaded plugin descriptions the class ompl_interface/OMPLPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are Available plugins: 
[move_group-7] [INFO] [1737563938.234976953] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.path_tolerance' was not set. Using default value: 0.100000
[move_group-7] [INFO] [1737563938.234993947] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.resample_dt' was not set. Using default value: 0.100000
[move_group-7] [INFO] [1737563938.235001214] [moveit_ros.add_time_optimal_parameterization]: Param 'ompl.min_angle_change' was not set. Using default value: 0.001000
[move_group-7] [INFO] [1737563938.235026809] [moveit_ros.fix_workspace_bounds]: Param 'ompl.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-7] [INFO] [1737563938.235045327] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_bounds_error' was set to 0.100000
[move_group-7] [INFO] [1737563938.235053509] [moveit_ros.fix_start_state_bounds]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-7] [INFO] [1737563938.235068348] [moveit_ros.fix_start_state_collision]: Param 'ompl.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-7] [INFO] [1737563938.235075461] [moveit_ros.fix_start_state_collision]: Param 'ompl.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-7] [INFO] [1737563938.235082439] [moveit_ros.fix_start_state_collision]: Param 'ompl.max_sampling_attempts' was not set. Using default value: 100
[move_group-7] [INFO] [1737563938.235096008] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-7] [INFO] [1737563938.235101869] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[move_group-7] [INFO] [1737563938.235106676] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-7] [INFO] [1737563938.235141209] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-7] [INFO] [1737563938.235147615] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[move_group-7] [INFO] [1737563938.235152270] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[move_group-7] [ERROR] [1737563938.237527525] [moveit.ros_planning_interface.moveit_cpp]: Failed to initialize planning pipeline 'ompl'.
[move_group-7] [INFO] [1737563938.238515001] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'chomp'
[... 107 lines cut ...]
[ros2_control_node-8] [INFO] [1737563948.934067725] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2_control_node-8] [INFO] [1737563949.030356434] [controller_manager]: Configuring controller 'dsr_moveit_controller'
[ros2_control_node-8] [INFO] [1737563949.030655204] [dsr_moveit_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[... cut ...] default_plugins/Odometry rviz_default_plugins/Path rviz_default_plugins/PointCloud rviz_default_plugins/PointCloud2 rviz_default_plugins/PointStamped rviz_default_plugins/Polygon rviz_default_plugins/Pose rviz_default_plugins/PoseArray rviz_default_plugins/PoseWithCovariance rviz_default_plugins/Range rviz_default_plugins/RelativeHumidity rviz_default_plugins/RobotModel rviz_default_plugins/TF rviz_default_plugins/Temperature rviz_default_plugins/TwistStamped rviz_default_plugins/Wrench

[image]

doosan-robotics, 2025-01-23:
안녕하세요 @hanprd 님

해당 현상은 moveit 패키지가 완전히 다 설치 되어있지 않아 발생하는 현상으로 보입니다.
`sudo apt-get install ros-humble-moveit*`

상위 커멘트로 moveit 플러그인을 설치하여 진행 부탁드립니다.

감사합니다.

hanprd, 2025-01-23:
> ---- External Email: Use caution with attachments, links, or sharing data ----

해결되었습니다.
감사합니다!

closed 2025-01-24 by doosan-robotics