DoosanRobotics/doosan-robot2 #162: How to send a packaging position without violating the joint limits

bxtbold, 2025-02-10:


leeminju531, 2025-02-10:
Hi, @bxtbold

Did you run it with the MoveIt launch? If so, you might need to modify the `joint_limit.yam`l file of your model (refer to [this](https://github.com/doosan-robotics/doosan-robot2/blob/humble-devel/dsr_moveit2/m1013_moveit_config/config/joint_limits.yaml)). 
Otherwise, please provide more details.

bxtbold, 2025-02-11:
Hi @leeminju531, thanks for a reply!

I am not using moveit but instead using `dsr_msgs2/srv/MoveJoint` service. Basically, what I was doing was requesting the below value to move the robot to the packaging position `[0.0, 0.0, 160.0, 0.0, 20.0, 0.0]`.

```
ros2 service call /dsr01/motion/move_joint dsr_msgs2/srv/MoveJoint "{pos: [0.0, 0.0, 160.0, 0.0, 20.0, 0.0], vel: 0.0, acc: 0.0, time: 2.0, radius: 0.0, mode: 0, blend_type: 0, sync_type: 0}"
```

The srv call resulted in something like `160 is beyond the limit of 135` from `dsr_hw_interface2` node.

I don't have an access to a doosan robot atm. Hence, I cannot provide the log. Maybe I will provide if needed later.

All in all, I just want to move the robot to its packaging position (without the teaching pendant).

doosan-robotics, 2025-02-11:
@bxtbold 

packing position is inherently a posture where the robot’s body parts can collide with each other. As a result, it is classified as a state that exceeds the robot’s safety limit. (Even on the Teaching Pendant, entering Recovery Mode and verifying safety status is required to access this position.)
Due to safety concerns, ROS currently does not support H2R-related commands for moving to the packing position.

🔹 Alternative Solutions
If you need to remotely move to the packing position (165°) using general motion, there are two possible solutions:

1. Adjust the Robot Limit Settings
Modify the joint angle limit on the TP screen, changing it from -145145 (default) to -165165 during the initial setup.
After adjusting the limits, issue the command via ROS again.
- Important Note:
If additional tools (e.g., end-effectors) are attached, there is a risk of self-collision, which may trigger collision detection and result in an STO (Safe Torque Off) state. Please use with caution.

2.Use Recovery Mode to Move the Joint
If necessary, you can enter Recovery Mode and move the joint to the desired position using the following steps:

servo_off 
-> set_safety_mode (Recovery Mode, enter)  
-> set_safety_mode (Recovery Mode, move)  
-> servo_on  
-> movej  

⚠️ Critical Considerations:
If the robot exits Recovery Mode while in the packing position, it will not be able to return to normal Auto Mode or Manual Mode.
Solution: After reaching the packing position, move back to within the -145~145 range before switching to Manual Mode and resuming operation.
In Recovery Mode, safety functions such as collision detection are disabled, increasing safety risks.
Use Recovery Mode only when absolutely necessary, as it bypasses safety features.

🚨 Final Reminder
For safety reasons, we strongly discourage using motion in Recovery Mode except for special cases like this. If you proceed, exercise extreme caution.

Thank you.

song-ms, 2025-02-11:
@bxtbold 

packing position is inherently a posture where the robot’s body parts can collide with each other. As a result, it is classified as a state that exceeds the robot’s safety limit. (Even on the Teaching Pendant, entering Recovery Mode and verifying safety status is required to access this position.)
Due to safety concerns, ROS currently does not support H2R-related commands for moving to the packing position.

🔹 Alternative Solutions
If you need to remotely move to the packing position (165°) using general motion, there are two possible solutions:

1. Adjust the Robot Limit Settings
Modify the joint angle limit on the TP screen, changing it from -145145 (default) to -165165 during the initial setup.
After adjusting the limits, issue the command via ROS again.
- Important Note:
If additional tools (e.g., end-effectors) are attached, there is a risk of self-collision, which may trigger collision detection and result in an STO (Safe Torque Off) state. Please use with caution.

2.Use Recovery Mode to Move the Joint
If necessary, you can enter Recovery Mode and move the joint to the desired position using the following steps:

servo_off 
-> set_safety_mode (Recovery Mode, enter)  
-> set_safety_mode (Recovery Mode, move)  
-> servo_on  
-> movej  

⚠️ Critical Considerations:
If the robot exits Recovery Mode while in the packing position, it will not be able to return to normal Auto Mode or Manual Mode.
Solution: After reaching the packing position, move back to within the -145~145 range before switching to Manual Mode and resuming operation.
In Recovery Mode, safety functions such as collision detection are disabled, increasing safety risks.
Use Recovery Mode only when absolutely necessary, as it bypasses safety features.

🚨 Final Reminder
For safety reasons, we strongly discourage using motion in Recovery Mode except for special cases like this. If you proceed, exercise extreme caution.

Thank you.

doosan-robotics, 2025-02-19:
If there are any further issues, please provide an update.
Otherwise, this issue will be closed.

closed 2025-02-19 by doosan-robotics