DoosanRobotics/doosan-robot2 #167: joint error wiith M1013

hvtung26, 2025-02-21:
I am using moveit2 for planning. It's work well except these two errors just keep continously pop up so i cannot see any of debuging. Could u pls let me know how should I solve it
[move_group-7] [ERROR] [1740128996.512909127] [moveit_robot_model.robot_model]: Joint 'steering_rr_joint' not found in model 'm1013'
[move_group-7] [ERROR] [1740128996.512911460] [moveit_robot_model.robot_model]: Joint 'wheel_fl_joint' not found in model 'm1013'

song-ms, 2025-02-24:
Hello @hvtung26 ,

We don't have any definition 'wheel' in Moveit2 command on Humble verison.
Could you let me knwo what command you used for moveit2 ?

hvtung26, 2025-02-24:
Hello,
I am using **ros2 launch dsr_bringup2 dsr_bringup2_moveit.launch.py mode:=real model:=m1013 host:=192.168.137.100**
and this is  a part of my logs:
ubuntu@ubuntu-16ZD90SP-EX79K:~$ ros2 launch dsr_bringup2 dsr_bringup2_moveit.launch.py mode:=real model:=m1013 host:=192.168.137.100
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-02-24-11-59-19-492094-ubuntu-16ZD90SP-EX79K-27125
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [set_config-1]: process started with pid [27127]
[INFO] [run_emulator-2]: process started with pid [27129]
[INFO] [robot_state_publisher-3]: process started with pid [27131]
[INFO] [spawner-4]: process started with pid [27133]
[INFO] [spawner-5]: process started with pid [27135]
[INFO] [spawner-6]: process started with pid [27137]
[INFO] [move_group-7]: process started with pid [27139]
[robot_state_publisher-3] [INFO] [1740365959.783659538] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1740365959.783785690] [robot_state_publisher]: got segment link_1
[robot_state_publisher-3] [INFO] [1740365959.783791809] [robot_state_publisher]: got segment link_2
[robot_state_publisher-3] [INFO] [1740365959.783795367] [robot_state_publisher]: got segment link_3
[robot_state_publisher-3] [INFO] [1740365959.783798398] [robot_state_publisher]: got segment link_4
[robot_state_publisher-3] [INFO] [1740365959.783801485] [robot_state_publisher]: got segment link_5
[robot_state_publisher-3] [INFO] [1740365959.783804782] [robot_state_publisher]: got segment link_6
[robot_state_publisher-3] [INFO] [1740365959.783807915] [robot_state_publisher]: got segment world
[move_group-7] Error:   Semantic description is not specified for the same robot as the URDF
[move_group-7]          at line 681 in ./src/model.cpp
[move_group-7] [INFO] [1740365959.784966931] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0277164 seconds
[move_group-7] [INFO] [1740365959.785020723] [moveit_robot_model.robot_model]: Loading robot model 'm1013'...
[run_emulator-2] Run Emulator of the Doosan Robot Controller
[run_emulator-2] dirname: /home/ubuntu/ros2_ws/install/common2/share/common2/bin/run_drcf.sh
[run_emulator-2] server_port: 12345
[run_emulator-2] robot model: M1013 
[run_emulator-2] ns: 
[run_emulator-2] run : 'emulator' ...
[run_emulator-2] permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get "http://%2Fvar%2Frun%2Fdocker.sock/v1.48/containers/json?filters=%7B%22name%22%3A%7B%22emulator%22%3Atrue%7D%7D": dial unix /var/run/docker.sock: connect: permission denied
[run_emulator-2] docker: permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Head "http://%2Fvar%2Frun%2Fdocker.sock/_ping": dial unix /var/run/docker.sock: connect: permission denied
[run_emulator-2] 
[run_emulator-2] Run 'docker run --help' for more information
[run_emulator-2] ARCH: 64-bit
[move_group-7] [WARN] [1740365959.935373909] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[spawner-6] [INFO] [1740365959.955237073] [spawner_dsr_moveit_controller]: waiting for service /controller_manager/list_controllers to become available...
[spawner-4] [INFO] [1740365959.960414764] [spawner_dsr_controller2]: waiting for service /controller_manager/list_controllers to become available...
[INFO] [set_config-1]: process has finished cleanly [pid 27127]
[spawner-5] [INFO] [1740365960.016395799] [spawner_joint_state_broadcaster]: waiting for service /controller_manager/list_controllers to become available...
[INFO] [ros2_control_node-8]: process started with pid [27235]
[ros2_control_node-8] [WARN] [1740365960.147425662] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-8] [INFO] [1740365960.147639059] [resource_manager]: Loading hardware 'm1013' 
[ros2_control_node-8] [INFO] [1740365960.152942907] [resource_manager]: Initialize hardware 'm1013' 
[ros2_control_node-8] [INFO] [1740365960.153129376] [dsr_hw_interface2]: [on_init] Hardware name : m1013, type : system, class type : dsr_hardware2/DRHWInterface
[ros2_control_node-8] [INFO] [1740365960.159904106] [dsr_hw_interface2]: name: 
[ros2_control_node-8] [INFO] [1740365960.159941857] [dsr_hw_interface2]: rate: 100
[ros2_control_node-8] [INFO] [1740365960.159957644] [dsr_hw_interface2]: standby: 5000
[ros2_control_node-8] [INFO] [1740365960.159965014] [dsr_hw_interface2]: command: true
[ros2_control_node-8] [INFO] [1740365960.159968753] [dsr_hw_interface2]: host: 192.168.137.100
[ros2_control_node-8] [INFO] [1740365960.159973629] [dsr_hw_interface2]: port: 12345
[ros2_control_node-8] [INFO] [1740365960.159977831] [dsr_hw_interface2]: mode: real
[ros2_control_node-8] [INFO] [1740365960.159992200] [dsr_hw_interface2]: model: m1013
[ros2_control_node-8] [INFO] [1740365960.159997601] [dsr_hw_interface2]: gripper: none
[ros2_control_node-8] [INFO] [1740365960.160012516] [dsr_hw_interface2]: mobile: none
[ros2_control_node-8] [INFO] [1740365960.160020352] [dsr_hw_interface2]: rt_host: 192.168.137.50
[ros2_control_node-8] [INFO] [1740365960.160070186] [dsr_hw_interface2]: _______________________________________________
[... cut ...]
[ERROR] [ros2_control_node-8]: process has died [pid 27235, exit code -6, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args -r __ns:=/ --params-file /tmp/launch_params_6_3xz28t --params-file /home/ubuntu/ros2_ws/install/dsr_controller2/share/dsr_controller2/config/dsr_controller2.yaml'].
[ERROR] [move_group-7]: process has died [pid 27139, exit code -11, cmd '/opt/ros/humble/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_1cu6u11z'].
[move_group-7]

doosan-robotics, 2025-02-25:
@hvtung26 

We could not find the issue on our site during testing.
Please delete the build/install log files in your workspace and perform a clean rebuild from source before trying again. 
Let us know if the issue persists.

hvtung26, 2025-02-25:
Hello, after rebuilding my ros2. Seems like the error does not exist at all. Thank you for your supporting

closed 2025-02-25 by doosan-robotics