DoosanRobotics/doosan-robot2 #172: warning Alarm Info: group(5), index(7056), param(), param(), param() while moving with c++ windows.

hvtung26, 2025-02-25:
Hello, I am using M1013, and the error Alarm Info: group(5), index(7056), param(), param(), param() appear anytime when the robot moves with a command. Even go home or jog or movel, but it works well with the same position when I control using a pendant. Could you please show me what should I do in my case?

leeminju531, 2025-02-26:
Hi, @hvtung26 

First of all, regarding https://github.com/doosan-robotics/doosan-robot2/issues/93#issuecomment-2680255244, did you run it on linux where there were no issue, but the problem occurs on windows ? 

For further investigation, Could you provide the exported logs from our controller after reproducing the situtation ? 
You can refer to https://github.com/doosan-robotics/doosan-robot2/issues/149#issuecomment-2594287513 for guidance.

Best regards,

hvtung26, 2025-02-28:
Yes, that's corrected. Before I try on Linux, there's no problem on Windows version. But after I tried on Linux and got back to Windows, the error occurred.  But I found a way to solve the problem: I set **Drfl.set_robot_mode(ROBOT_MODE_AUTONOMOUS);
Drfl.set_safety_mode(SAFETY_MODE_AUTONOMOUS, SAFETY_MODE_EVENT_MOVE);** before every movel() cmd I. Then the error disappears. I cannot find any log from the controller but it debug in my window console Alarm Info: group(5), index(7056), param(), param(), param().

song-ms, 2025-03-04:
Hello @hvtung26 ,

If you need to move more than 100 positions, the movesx command may not be suitable due to its position limit of 100. 
Here are two alternative solutions:

1. If you are using an integrated controller, upgrading to version 3.4 (scheduled for release in early May) will allow for extended motion control.
2. Alternatively, using movel with blending can enable smooth movement by sequentially commanding each target position.

hvtung26, 2025-07-30:
Hello, any information about the new release that support extended motion control?