DoosanRobotics/doosan-robot2 #177: Issue: Connection Failure in Virtual Mode for Doosan Robot M1013 on ROS 2 Humble (WSL2)

shimtaek, 2025-03-05:
Hello When I launching the Doosan Robot emulator (dsr_bringup2) in virtual mode, the connection repeatedly fails with the error "Connection refused". The ros2_control_node also fails to initialize hardware and eventually crashes.

- System Information

Robot Model: M1013
ROS 2 Distribution: Humble
OS: Ubuntu 22.04 on WSL2 (Windows Subsystem for Linux)

**- Launch Command**

ros2 launch dsr_bringup2 dsr_bringup2_rviz.launch.py mode:=virtual host:=127.0.0.1 port:=12345 model:=m1013

### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
**-Error Logs (Relevant Excerpts)-**
[ros2_control_node-6] Connection refused
[ros2_control_node-6] Connect Failed Please check network state : Connection refused
[ros2_control_node-6] [INFO] [dsr_hw_interface2]: Connecting failure.. retry...
...
[ros2_control_node-6] [ERROR] [dsr_hw_interface2]: DSRInterface::init() DRCF connecting ERROR!!!
[ros2_control_node-6] [INFO] [resource_manager]: Failed to initialize hardware 'm1013'
[ros2_control_node-6] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-6]   what():  Wrong state or command interface configuration.

### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### ### 
I HAVE A PROBLEM SAME AS GAZEBO 
ros2 launch dsr_bringup2 dsr_bringup2_gazebo.launch.py mode:=virtual host:=127.0.0.1 port:=12346 name:=dsr01 x:=0 y:=0

**### Error Logs (Relevant Excerpts)**
[ros2_control_node-7] [INFO] [dsr_hw_interface2]: Connecting failure.. retry...
[run_emulator-2] Run Emulator of the Doosan Robot Controller
[ros2_control_node-7] Connect Failed Please check network state : Connection refused
[gazebo_connection-3] process has died [pid 67970, exit code -2]
[run_emulator-2] process has died [pid 67968, exit code -2]

justhash-kr, 2025-03-06:
Hi. @shimtaek .

I resolved this issue by adding the `--privileged` option to the **`docker run` command** in the **`run_drcf.sh`** script. 

(I'm not entirely sure about the exact difference between WSL2 and a native Ubuntu environment, but I suspect this issue might be related to differences in kernel usage.)  

The `run_drcf.sh` file is located in **`doosan-robot2/dsr_common2/bin/`**.  

After making this change, you need to run **`colcon build`** again.

I hope this solution does works you too.

I've already reported this issue and the solution to the Doosan team, and they've confirmed that they will investigate and make the necessary updates.

leeminju531, 2025-03-12:
Dear, @shimtaek , @justhash-kr 

Thanks for reporting this. updated at https://github.com/doosan-robotics/doosan-robot2/pull/184

closed 2025-03-13 by doosan-robotics